prusa3d/PrusaSlicer #10441: Solid Infill Over Bridge Infill

dondec123, 2023-04-25:
### Description of the bug

In the included example project I show pic 1 - spare infill above bottom layers (reliable done super fast on my modern CoreXY).  
Next layer is pic 2 - bridge Infill I slow way down (40mms).  Print looks perfect so far.
Next layer PS jumps to Solid Infill pic 3 - (which on my machine that goes at150mms).

PROBLEM: The delicate bridge layer cannot survive that kind of speed increase (from 40mms to 150 over a bridge).

Modern printers are going faster and faster now and this is a basic Speed Consideration problem with PS.  

PS needs to start considering speed reductions for Solid Infill over Bridge Infill, or more conveniently for the user, slowly ramping up from the Bridge Infill speed.   For example:

1.	Allow user to specify what percent speed increase to use on the layers above a Bridge.  E.G., if Bridge speed is 40mms, make next layer of “Solid infill” to be 50mms (25% faster than 40), then 80mms next layer (100% faster than 40) … then back to full Solid Infill speed (for me at 150mms).

2.	Alternatively, but not the best, is to allow Height Range Modifiers that can RAMP SPEEDS upward over the Height Range.  This is only a “band aid” for the problem though, requires more work for the user to get it right, and therefore not as desirable as solution 1.
[image]
[image]
[image]

### Project file & How to reproduce

[Box 120x120 33 31 32 Faster.zip](https://github.com/prusa3d/PrusaSlicer/files/11316005/Box.120x120.33.31.32.Faster.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha 6

### Operating system

Win 11

### Printer model

Formbot Troodon 2.0 (voron 2.4 clone)

Jan-Soustruznik, 2023-05-17:
Hello, @dondec123 , 

Thank you for your feedback. I think that your problem may be solved by **Pressure equalizer** feature located in _Print settings -> Speed_. Please check also attached link on our Prusa knowledge base where you can find some basic information how this feature use, link: https://help.prusa3d.com/article/pressure-equalizer-_331504

Thank you, have a nice day

dondec123, 2023-05-18:
Hi Jan, No this is not a Pressure Equalizing thing.  What I’m talking about is:I have my Solid Infill setting set to 135mm (way fast), bc it works on a hard flat base.    BUTWhen the next layer is done over delicately extruded “BRIDGING over long air gap”, PS defaults to SOLID INFILL for that, and at 135mms it rips up that bridging up every time. What my ticket is suggesting is: There aught to be more granularity for this situation, another alternative for layers immediately extruded on top of fine bridging… not just blindly choosing Solid Infill.  Solid Infill SPEED is a bad choice for fast printers. Printers are getting much faster now, and there needs to be more granularity in speed controls for this situation.   Please again refer to my ticket.   Hope this helps.     Don From: Jan SoustružníkSent: Wednesday, May 17, 2023 9:50 AMTo: prusa3d/PrusaSlicerCc: dondec123; MentionSubject: Re: [prusa3d/PrusaSlicer] Solid Infill Over Bridge Infill (Issue #10441) Hello, @dondec123 ,Thank you for your feedback. I think that your problem may be solved by Pressure equalizer feature located in Print settings -> Speed. Please check also attached link on our Prusa knowledge base where you can find some basic information how this feature use, link: https://help.prusa3d.com/article/pressure-equalizer-_331504Thank you, have a nice day

closed 2026-09-29 by prusaslicer-bot[bot]