prusa3d/PrusaSlicer #10448: fail to print several models in completing individual objects mode

happydrones, 2023-04-26:
### Description of the bug

when i print several  same models  at ones  , even if there is enough distance between two  models  the printer still notice anerror showed in the picture below it show a error in below picture 
[image]

### Project file & How to reproduce

[2.zip](https://github.com/prusa3d/PrusaSlicer/files/11334083/2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2+win64

### Operating system

win 10

### Printer model

creality cr

happydrones, 2023-04-26:
printer model is creality cr-10 smart

neophyl, 2023-04-26:
Im sorry but a stl file is not any use.  What is needed is your project file saved from Prusa Slicer.  Get it to the point where the error is generated and then use FIle>Save Project as.  That generates a PS .3mf project file.  It will contain the model but more importantly a copy of ALL of your settings.  

It will be the settings that is causing the warning.
For example with my Cr10 Settings I can slice 2 of your stl's
[image]

sarusani, 2023-04-26:
If you want to print tall objects like that, you can't use sequential printing. Turn it off or print them in two separate jobs.

neophyl, 2023-04-26:
You can if you are careful and plan it out in combination with the right settings configured in your print profile though.  For a 'normal' bedslinger design you print from front to back.  That way the height between the nozzle and X axis isnt a factor, just the footprint around it.  

Of course you as the user are then responsible for making sure none of the path planning will intersect previous printed models, especially things like end gcode etc.  As you can reorder the parts in the right tree window it becomes quite possible.

happydrones, 2023-04-27:
[2.zip](https://github.com/prusa3d/PrusaSlicer/files/11340772/2.zip)
this new zip file  contains a .3mf project file

sarusani, 2023-04-27:
>For a 'normal' bedslinger design you print from front to back.
Depends on the printer and firmware you're using.
On a Mk3 for example, only if you disable crash detection, power failure recovery and don't use m600 color changes etc. All of these cases move the head to the front. During the printing of the 2nd, 3rd... objects.

But obviously OP doesn't have theses settings adjusted, so as said, he can't print objects that tall with sequential printing enabled.

Jan-Soustruznik, 2023-05-29:
Hello, @happydrones , 

If correctly understood your problem is, that is not possible to slice your project when you enable the "Sequential printing".

This will be problem that the models have 66mm height and the "Sequential printing" height in default is set 25mm. So If you really know that the updating of setting will not damage your printer. Change the height to 67mm, then slicer will slice your project.

happydrones, 2025-02-28:
> Hello, @happydrones ,
>
> If correctly understood your problem is, that is not possible to slice your project when you enable the "Sequential printing".
>
> This will be problem that the models have 66mm [...]

I'm really sorry for responding so late, and I also found the same solution. Thank you very much for your reply, I appreciate it.

closed 2025-03-01 by happydrones