prusa3d/PrusaSlicer #10454: After filament change, printer extrudes extra filament that causes an erroneous bump in the surface

puterboy, 2023-04-27:
### Description of the bug

After a filament change is completed with the old filament flushed and (manually) wiped away, the extruder proceeds to extrude about 10-15mm of additional filament.

This filament simply dangles (like a piece of loose spaghetti) from the extruder and then gets laid down when the extruder head then descends at the first place where the extruder crosses the object.

The only solution is to immediately intervene and manually wipe this extra (unnecessary) extruded noodle before the extruder head resumes printing.

### Project file & How to reproduce

Reproduce by initiating/causing a filament change and watching what happens after the filament change is confirmed on the keyboard.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6

### Operating system

Windows 11

### Printer model

Prusa MK3S+

puterboy, 2023-04-27:
I saw that some people claim this is to "repressurize" the extruder.
But why does the extruder need to be "repressurized" since by definition you just changed the film which primed the extruder with new filament. Moreover, you just confirmed that it is extruding properly by clicking "yes".

Perhaps there is a very rare edge case where somebody returns to the printer to change the filament, removes the old filament, inserts a new one and then walks away for a while before coming back to press "yes". But that would seem to be a rare case as presumably most people press "yes" soon as they see the filament has extruded properly -- after all they WANT the print to resume. 

Also, if this were a major problem, then why doesn't the extruder repressurize whenever you manually "pause" the print. In that case the time to "depressurize" can be very long and even worse the hotend is allowed to cool.

puterboy, 2023-04-27:
If you still want to allow for repressurization, why not add this as an option under either "Print Settings" or "Printer Settings". The option would be to set the LENGTH of the extra extrusion. Setting to '0' would give no extra extrusion. Setting to 5mm would give the current behavior.

neophyl, 2023-04-27:
As mentioned on the prusa forum this is down to printer firmware and not Prusa Slicer.  

See issue https://github.com/prusa3d/Prusa-Firmware/issues/197

closed 2023-04-27 by rtyr