prusa3d/PrusaSlicer #10455: Add option to call external GCODE post-processing script.

puterboy, 2023-04-27:
Often, users may need to post-process their GCODE to add changes that go beyond the options included within Prusa Slicer. This would typically be a script such as a Python/Perl/Shell script that parses and regex match/replaces/adds various elements to the GCODE.

Now if you are downloading the GCODE, you can always run such a script manually - though it is a pain to do so or even remember to do so. But typically, I upload the GCODE directly from Prusa Slicer to my printer which gives no opportunity to post-process the GCODE.

So my suggestion is to allow for a filter-type script to be called after the GCODE is created. This would appear as an optional setting allowing users to add (or browse to) the path of the script.

Ideally, such an option would be included (separately)  as an advanced feature under "Print Settings", "Filament Settings" and "Printer Settings" since there might be some scripts that are print-dependent, others that are filament-dependent, and still others that are printer-dependent.

This would seem to be an almost trivial feature to add.

puterboy, 2023-04-27:
I just noticed that such an option exists under 'expert' mode under "Print Settings" -- so that is good.
Still would be helpful to have similar options under "Filament Settings" and "Printer Settings" -- since as noted above your need for post-processing can sometimes be print dependent, other times filament-dependent, and other times printer-dependent.

zbrozek, 2023-12-16:
+1 to adding this option to `printer_settings` and possibly also `filament_settings`. I'd like to use `klipper_estimator` and don't have a good way to use the correct config settings per-printer.

foreachthing, 2023-12-16:
Why don't you add a note to your filament (or something else, like start gcode) and check for that in your post-processing script? That way you only have to maintain _one_ script.

zbrozek, 2023-12-16:
- I don't write `klipper_estimator` and would have to wrap it in another script to parse the gcode file and then call the next script.
- It's not really a clear or elegant solution since it's adding additional layers of indirection.

That's a viable workaround, but I don't plan to use it.

puterboy, 2026-09-02:
Keep this alive until fixed!