prusa3d/PrusaSlicer #10461: Prusa Currency

PublicReserveBank, 2023-04-28:
### Description of the bug

so my problem is with currency like USD, PLN, EUR

please add this i want this so much, prusa is better than cura slicer with this you have better things but add this and its better

### Project file & How to reproduce

no

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 a6

### Operating system

win 10

### Printer model

ender3pro no

Jan-Soustruznik, 2023-05-18:
Hello, @PublicReserveBank , 

Could you more specify your request, please?

PublicReserveBank, 2023-05-21:
yes, make better place for currency, i made a program with bing ai to
calculate in 5 minutes, and you have whole app and cant bruh, usd at least,
if you want i can help

kubispe1, 2023-05-22:
`yes, make better place for currency, i made a program with bing ai to
calculate in 5 minutes, and you have whole app and cant bruh, usd at least,
if you want i can help`

If you have a special idea, PS is open source, so feel free to offer a pull request. Then we can accept it or, at the very least, be inspired and cooperate on your solution.
Thanks.

rtyr, 2023-12-02:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2563

closed 2023-12-02 by rtyr