prusa3d/PrusaSlicer #10464: 2.6.0-alpha6: organic supports print through other parts on buildplate.

BakeJake, 2023-04-29:
### Description of the bug

Organic supports seem to ignore other parts on the buildplate and print through them.
[image]

### Project file & How to reproduce

[AquariumPlantHanger.zip](https://github.com/prusa3d/PrusaSlicer/files/11359074/AquariumPlantHanger.zip)
Set Body19 to 100% rectilinear, 30% gyroid on body1, supports on buildplate only: organic

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6

### Operating system

Windows 11

### Printer model

Creality Ender-3 S1 On Sonic Pad

foreachthing, 2023-04-29:
https://github.com/prusa3d/PrusaSlicer/issues?q=supports+through+objects+

@ moderators: pin topic?

BakeJake, 2023-04-29:
> https://github.com/prusa3d/PrusaSlicer/issues?q=supports+through+objects+
> 
> @ moderators: pin topic?

I'm actually sorry It was like 3am and I somehow missed all the other reports on this, and seems its prusaslicer in general as it treats objects as individuals and slices as such

foreachthing, 2023-04-30:
No need to apologize :-) 
imho: Prusa-admins/moderators should make this "issue" more prominent.

I think Prusa is discussing this topic, since a lot of (new?) users are having issues how PS works.
Once you know the "trick", it's easy to work with.

kubispe1, 2023-05-02:
We are trying to put together at least SOME solution due to an enormous amount of issues. 
Closing this issue. #316

closed 2023-05-02 by kubispe1