prusa3d/PrusaSlicer #10467: Start G-Code cannot resolve temperature variables

blast-hardcheese, 2023-04-30:
### Description of the bug

Using `M190 S[first_layer_bed_temerature]` or `M109 S[first_layer_temperature]` in `Custom G-Code` > `Start G-Code` results in the following error:

```
G-code export to /var/folders/9y/yxxwrfsx1gn2lntw9qnq_lv40000gn/T/.59969.gcode failed due to invalid custom G-code sections:
start_gcode
Parsing error at line 7: Variable does not exist
M190 S[first_layer_bed_temerature] ; set bed temperature and wait for it to be reached
       ^
Please inspect the file /var/folders/9y/yxxwrfsx1gn2lntw9qnq_lv40000gn/T/.59969.gcode.tmp for error messages enclosed between
        !!!!! Failed to process the custom G-code template ...
and
        !!!!! End of an error report for the custom G-code template ...
for all macro processing errors.
```

I've also tried `M109 S{first_layer_temperature}` but that results in the same or a similar error.

### Project file & How to reproduce

Put the following in your `Custom G-Code` > `Start G-Code`, then attempt to slice:

```gcode
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

;; Start Purge Line

M107
M190 S[first_layer_bed_temerature] ; set bed temperature and wait for it to be reached

G92 E0  ; Set extruder offset
G1 Y-2 X179 F2400  ; Prepare for purge line
G1 Z3 F720
M109 S[first_layer_temperature] ; wait for extruder temp

; intro line
G1 X170 F1000
G1 Z0.2 F720
G1 X110 E8 F900
G1 X40 E10 F700
G92 E0
;; End Purge Line
```

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+MacOS-arm64

### Operating system

macOS Ventura, 13.1 (22C65)

### Printer model

Sovol SV06, stock

mix579, 2023-04-30:
Not claiming to be a gcode expert but first_layer_bed_temerature is clearly a type and should be first_layer_bed_temperature (with a p)

blast-hardcheese, 2023-04-30:
That turned out to be it. Sorry for the noise!

closed 2023-04-30 by blast-hardcheese

blast-hardcheese, 2023-04-30:
Walking through the steps I took again to figure out where I could have noticed this, the issue was that the search box does autocompletion for partial inputs, so it added the missing P when I searched for the variable directly. Having something similar before displaying the error would be ideal, but it's really a nice-to-have since clearly it works as is.