prusa3d/PrusaSlicer #10470: [Feature Request] Customizable extrusion multiplier (flow) value for the different aspects of the print (infill, solid top infill, perimeter, etc..)

StevenNitsuga, 2023-04-30:
**Is your feature request related to a problem? Please describe.**
I want to be able to adjust my infill flow rate just like in cura, as my nozzle keeps dragging on top of the infill due to over-extrusion on  infill. (Yes, I have calibrated every aspects of my printer, and found out that I could solve it if i could lower my flow on my infill in relation to my global extrusion multiplier)

**Describe the solution you'd like** **Describe how it would work**
The solution is to create an option where we can set the extrusion multiplier for infill, top solid infill, perimeter, supports, etc.. and not rely on a single global value for an extrusion multiplier.

**Additional context**
I found a feature request going back all the way from 2021, but still no update to it. I'm writing this request for the feature to be followed up.

kubispe1, 2023-05-02:
There are similar issues, so I link them together at least. #10280 and there is a kind of original #7200

riccalisci, 2023-05-03:
please add independent flow control, at least for the first layer!

arshish1612, 2023-07-18:
+ 1 When will this be added pls?

arshish1612, 2023-08-19:
Any update to when or if this will be implemented please?

Tupson444, 2026-09-18:
still relevant

Jan-Soustruznik, 2026-10-20:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16010.

closed 2026-10-20 by Jan-Soustruznik