prusa3d/PrusaSlicer #10471: App crashing when creating additional instance of a model that is hidden

kinematicdigit, 2023-04-30:
### Description of the bug

Minor issue when trying to make an additional instance of an object that was previously hidden. The solution is simply making it visible, but it is consistent for me and reproducible.

PrusaSlicer
Version:   2.6.0-alpha6+MacOS-arm64
Build:     PrusaSlicer-2.6.0-alpha6+MacOS-arm64-202303311518

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Ventura Version 13.2 (Build 22D49)
Total RAM size [MB]: 68,719MB
OpenGL installation
GL version:   4.1.0
Profile:      Core
Vendor:       Apple
Renderer:     Apple M1 Max
GLSL version: 4.10.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>

### Project file & How to reproduce

[circle_good.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11361752/circle_good.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+MacOS-arm64-202303311518

### Operating system

macOS Ventura Version 13.2 (Build 22D49)

### Printer model

Ender 3V2 with Sonic Pad

YuSanka, 2023-05-02:
Thanks for your report. Unfortunately, we can't reproduce this issue. Could you add some video with crash? Or at least 3mf-file?

kinematicdigit, 2023-05-03:
> 

3MF is already uploaded. 

If you have multiple items, hide one of them, then select it and try to make another instance of it. It should crash; I've had this happen in multiple instances and restarts of the application.

Jan-Soustruznik, 2023-05-17:
Hello, @kinematicdigit, 

Could you check with latest beta, please? 

Thank you, have a nice day

kinematicdigit, 2023-05-18:
Hi there,

Just downloaded the new version and indeed the bug is fixed in the new beta release.

Thank you!

—

Terrance Lam
Project Manager, Marketing and Designer
Danada Enterprises Ltd. | Columbia Energy | Enrico Winery | Accretive Holdings Inc.

2519 B Rockbay Ave | Victoria | BC | V8T 4R5
Mobile. 250.888.4942
Telephone. 250.478.1272

Jan-Soustruznik, 2023-05-29:
@kinematicdigit , Thank you for info. I closed issue as solved. Have a nice day

closed 2023-05-29 by Jan-Soustruznik