prusa3d/PrusaSlicer #10472: Thumbnail is not in one line within the gcode

TylonHH, 2023-05-01:
### Description of the bug

My printer doesnt recognized the thumbnail. So I looked into the gcode.
There is see this:
```
; thumbnail begin 200x200 20088
; iVBORw0KGgoAAAANSUhEUgAAAMgAAADICAYAAACtWK6eAAA6n0lEQVR4Ae2dWZAexbXnPzazSixikU
; ELEhLa96WlXrTvO2hBQgLtu4QksDH2tcE2NmBssLHxtQ02O9jGxgIM2GAwBswmi4iZiHmcibhv83Ij
; ZubxTsRM3frXV/l1VtbJzJNLVUu4H050q9X9LfXlr/7nf/JkZmPWrFkJFbNnz86ira0tmTBhQuvfVD
; QajezrnDlznGPu3LnWwOOL7+fNm0cGfof6+fz5851iwYIFpWhvb8+uA/V/iIULF2bPj68usWjRInbg
; 8XX/t3jx4izwO+L72IFrg+uwZMkSbZx99tnG/w8NvD+f3wl53w0dGPLAt4UY7GeeeWZUMBAXXHCBFg
; odJDGgEIHH1EFBhQ0UFyhEnHPOOSUYtB9oACQhAzPk9zlx/vnnV/L8bEDUQY/BzgVDDry4GGDoFMEU
; YsCfd955QVCIwB2RA4UJEh8gZNXwvvNp/tZ1YPoO9liQhDxOyN+2rqMuXXIFQ44ZM2YkEydO9ALDFQ
; 4dAHgMXzBk1XBNmwQU+Nvhw4c7QxFDCQQIPTW45OjXr1+PvobQa9BwgcPVX8jegRN9+/YNBkMOeCcu
...
```
So I removed all the line breaks and comments. Then I got all the code in one line. Saved the gcode and voilà. I see the thumbnail.
Even if I check the lines with the code into https://base64.guru/converter/decode/image/png I will see a thumbnail if the code is in one line. So why is there a line break?

### Project file & How to reproduce

Printer settings -> gcode Thumbnail 200x200 (Marlin) / PNG
Check out the gcode. It shouldnt be line breaked.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.2+win64

### Operating system

Win10

### Printer model

Snapmaker J1

kubispe1, 2023-05-02:
I'm afraid that the only solution is to request the developers of the Snapmaker J1 printer firmware to extend support for this syntax with `;` . The reason for this is that if we change it and ignore the `;`, it would be unreadable for many other firmwares and printers. 
To support my statement, the Cura post-processing built-in script uses a similar syntax with `;` .

closed 2023-05-02 by kubispe1

foreachthing, 2023-05-02:
@TylonHH, you could write a small post-processing script to merge those lines after `; thumbnail begin 200x200 20088` to "fix" it for your printer.