prusa3d/PrusaSlicer #10473: Scale against a measurement

mippen, 2023-05-01:
**Describe the solution you'd like**
I would like to scale a component so a measurement matches.
For example an embedded 10mm magnet.. And I would like to scale against a 12mm

**Describe how it would work**
Simpliest would be to see a measurement active while scaling as normal.
Would be great to be able to scale with setting a new dimension on a measurement.

kubispe1, 2023-05-02:
Hope I understand it clear, but it is implemented already. Use the measurement tool and edit by the right button.

[image]

mippen, 2023-05-02:
Can I lock the scaling to only affect x and y?

closed 2026-09-29 by prusaslicer-bot[bot]