prusa3d/PrusaSlicer #10475: Error during dependencies build

HornedHeck, 2023-05-02:
### Description of the bug

I'm trying to build PS from sources on Fedora 38. During `make` stage I encountered with next error:

```
[ 80%] Building CXX object OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfDeepCompositing.cpp.o
[ 80%] Building CXX object OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfFastHuf.cpp.o
[ 81%] Building CXX object OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfFloatVectorAttribute.cpp.o
[ 81%] Building CXX object OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfRle.cpp.o
[ 82%] Building CXX object OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfZip.cpp.o
[ 83%] Building CXX object OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfSystemSpecific.cpp.o
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfHuf.cpp: In function ‘void Imf_2_5::hufUncompress(const char*, int, short unsigned int*, int)’:
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfHuf.cpp:1105:5: error: ‘uint64_t’ was not declared in this scope
 1105 |     uint64_t nBytes = (static_cast<uint64_t>(nBits)+7) / 8 ;
      |     ^~~~~~~~
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfHuf.cpp:56:1: note: ‘uint64_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
   55 | #include <algorithm>
  +++ |+#include <cstdint>
   56 | 
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfHuf.cpp:1107:16: error: ‘nBytes’ was not declared in this scope; did you mean ‘nBits’?
 1107 |     if ( ptr + nBytes > compressed+nCompressed)
      |                ^~~~~~
      |                nBits
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfHuf.cpp:1124:31: error: ‘nBytes’ was not declared in this scope; did you mean ‘nBits’?
 1124 |         if( ptr-compressed  + nBytes > nCompressed)
      |                               ^~~~~~
      |                               nBits
gmake[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/build.make:328: OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfHuf.cpp.o] Error 1
gmake[5]: *** Waiting for unfinished jobs....
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfMisc.cpp:1392:17: error: ‘uint8_t’ does not name a type
 1392 | struct FBytes { uint8_t b[4]; };
      |                 ^~~~~~~
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfMisc.cpp:55:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
   54 | #include <ImfTileDescription.h>
  +++ |+#include <cstdint>
   55 | #include "ImfNamespace.h"
gmake[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/build.make:468: OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfMisc.cpp.o] Error 1
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfDwaCompressor.cpp: In member function ‘void Imf_2_5::DwaCompressor::LossyDctDecoderBase::execute()’:
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfDwaCompressor.cpp:746:31: error: ‘uintptr_t’ does not name a type
  746 |         if ((reinterpret_cast<uintptr_t>(rowBlockHandle + i) & _SSE_ALIGNMENT_MASK) == 0)
      |                               ^~~~~~~~~
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfDwaCompressor.cpp:177:1: note: ‘uintptr_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
  176 | #include "dwaLookups.h"
  +++ |+#include <cstdint>
  177 | 
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfDwaCompressor.cpp:1014:38: error: ‘uintptr_t’ does not name a type
 1014 |                 if (reinterpret_cast<uintptr_t>(_rowPtrs[comp][y]) & _SSE_ALIGNMENT_MASK)
      |                                      ^~~~~~~~~
/home/hornedheck/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR/OpenEXR/IlmImf/ImfDwaCompressor.cpp:1014:38: note: ‘uintptr_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
gmake[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/build.make:454: OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfDwaCompressor.cpp.o] Error 1
gmake[4]: *** [CMakeFiles/Makefile2:444: OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/all] Error 2
gmake[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/dep_OpenEXR.dir/build.make:86: dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:369: CMakeFiles/dep_OpenEXR.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

### Project file & How to reproduce

git clone https://www.github.com/prusa3d/PrusaSlicer
cd PrusaSlicer
cd deps
mkdir build
cd build
cmake .. -DDEP_WX_GTK3=ON
make

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.2, master

### Operating system

Fedora 38

### Printer model

-

Salamandar, 2023-05-13:
Confirmed. This is due to the GCC 13 release. Other projects are broken as well. https://github.com/JuliaLang/julia/issues/49581

Salamandar, 2023-05-13:
FYI upstream OpenEXR builds properly (3.0)

closed 2026-09-29 by prusaslicer-bot[bot]