prusa3d/PrusaSlicer #10478: Bridging angle

JackDaniels223, 2023-05-03:
### Description of the bug

Bridging angle changes much too rapidly. 
I expected 1 degree at a time, but it's changing by MUCH more than 1 degree. 

### Project file & How to reproduce

Open Project, slice file. Drop down to 8.2 layer height to see the bridge. 
It's set at 15* here in my example. 

Change degree by going to Print Settings>Infill>Advanced>Bridging angle. And increase by 1. In my example 16*.

[error.zip](https://github.com/prusa3d/PrusaSlicer/files/11378834/error.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6-alpha5

### Operating system

Windows 10

### Printer model

MK3s+

neophyl, 2023-05-03:
https://github.com/prusa3d/PrusaSlicer/issues/9963  

Bridging angle has a known issue in the alpha 5 version you are using.  Its in Radians and not degrees.  Alpha 6 is supposed to have a fix in for it.

Jan-Soustruznik, 2023-05-18:
Know issue in a5, fixed in a6, 

Closed as solved

closed 2023-05-18 by Jan-Soustruznik