prusa3d/PrusaSlicer #10479: New Infill Type - Gradually Increasing Gyroid Infill Pattern

nxp200, 2023-05-03:
As an avid 3D printing enthusiast, I would like to propose the addition of a new infill pattern to PrusaSlicer, which gradually increases the infill percentage from the bottom to the top of the printed object. This new pattern would be based on the existing Gyroid infill pattern, with modifications to the infill percentage to create a smoother and more uniform structure throughout the printed object.

I have tried to use other infill types, but I found that they either use up too much unnecessary infill filament or, if not enough infill percentage is used, the first few top infill layers don't have enough to bridge and curl up. This has led to unsightly prints with poor structural integrity.

The proposed infill pattern would be ideal for creating parts with varying degrees of structural complexity. The lower layers of the infill would be set to a lower percentage, allowing for faster print times and reduced material usage while maintaining structural integrity. As the layers progress towards the top, the infill percentage would gradually increase, resulting in a more uniform and consistent structure with improved strength and durability. This feature would be especially beneficial for flat top surfaces where the first perimeter layer has more infill surface to sit on top of, resulting in a smoother surface finish.

I believe that the addition of this new infill pattern to PrusaSlicer would greatly benefit the 3D printing community and enhance the software's capabilities. Thank you for your consideration.

Jan-Soustruznik, 2023-05-18:
Hello, @nxp200 , 

What about Adaptive cubic infill?

> The Adaptive cubic infill works on the same principle as cubic. It consists of cubes oriented corner-down where lines cross themselves in one layer. Unlike simple cubic infill, this infill gets automatically more or less dense, depending on the distance to the nearest wall, leaving large cavities in the middle. This is especially useful for large prints with a big internal volume. The print will have denser infill at the top, bottom and around all sides, but it will have lower density near the center of the volume. The result is shorter print time and lower filament consumption while maintaining great support for top layers and similar mechanical properties. Material consumption is approx. 1/4 less than the rectilinear infill.

Or Support cubic infill?

> This infill gets automatically denser depending on the distance to the nearest top layer (the infill density increases only in the Z-axis). ts primary function is to support top layers by saving as much material as possible, it doesn’t improve any mechanical qualities of the model. Material consumption and print time of this infill are by far the lowest of all the supported infills.

source: https://help.prusa3d.com/article/infill-patterns_177130

nxp200, 2023-05-21:
Thanks Jan-Soustruznik! I think Support Cubic infill is what I needed. Don't know why I didn't see this before, but it does resolve my problem. Although, I still do think it would be nice to have Gyroid infill with the same effect.

Appreciate your time and help.

closed 2023-05-21 by nxp200