prusa3d/PrusaSlicer #10480: I have a suggestion. We will be glad if anyone is interested.

nazcak, 2023-05-03:
Hello. we know that material consumption and time are important in such programs.
I have a suggestion. We will be glad if anyone is interested.
  print settings ----> skirt and brim ---> brim .....
                      I thought of a great feature to add.
As it is known, the skirt unit is to ensure the adhesion of the material to the table.
but the time and material spent for this case are important. therefore, adhesion solutions can sometimes be made that are not required or with less material and time.
    RECOMMENDATION_1
-Brim section; For parts with a wide base surface, ADD UNITS line, such as circle, square, etc... ONLY to the corners of the parts, option.
    RECOMMENDATION_2
-Brim section; ADD extra units to the corners can be added to the Brim Type section.

[image]

[image]

neophyl, 2023-05-03:
Sounds like a duplicate of several existing feature requests, such as https://github.com/prusa3d/PrusaSlicer/issues/6208 or https://github.com/prusa3d/PrusaSlicer/issues/8283 fir example.

You can use modifiers currently to add small 'helper discs' in specific areas for troublesome parts but its a manual process.  Search for the term 'mouse ear'  .

Snuff1eupagus, 2024-04-27:
@nazcak try this guys method, it may help you.

https://www.youtube.com/watch?v=dhsbMmosbLU

hapiel, 2025-01-19:
> [@nazcak](https://github.com/nazcak) try this guys method, it may help you.
> 
> https://www.youtube.com/watch?v=dhsbMmosbLU

Link no longer works, but here is a new one: https://www.youtube.com/watch?v=IEKUpZ-gjMs

closed 2026-09-29 by prusaslicer-bot[bot]