prusa3d/PrusaSlicer #10484: Access Violation

carlWuestenberg, 2023-05-03:
### Description of the bug

Cannot slice project. Error message Access vailation. Objects placed on the build plate using the Arrage function. When sliced with Grid Support it works, when sliced with Organic it doesn't. Support on build plate only. 

### Project file & How to reproduce

[Full assembly (for FDM printers).zip](https://github.com/prusa3d/PrusaSlicer/files/11389817/Full.assembly.for.FDM.printers.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Windows 11

### Printer model

Prusa MK3S+

Jan-Soustruznik, 2023-05-04:
Hello, @carlWuestenberg, could you check with latest [alpha6,](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6) please?

lukasmatena, 2023-11-12:
No response, no other similar issue reported. Closing.

closed 2023-11-12 by lukasmatena