prusa3d/PrusaSlicer #10486: Organic supports intersect with geometry of neighbouring objects

VorpalBlade, 2023-05-03:
### Description of the bug

If two objects are close on the build plate (as placed by the built in arrange button) and organic supports are used, the supports may end up intersecting the neighbouring object.

This is hopefully not intended behaviour: PrusaSlicer should at least warn in this case, or if organic supports are enabled and there would be an intersection, arrange should place items further apart 

### Project file & How to reproduce

Here is the project file: [case.zip](https://github.com/prusa3d/PrusaSlicer/files/11391005/case.zip)

1. I used the "arrange" button to arrange some items on the build plate
2. I enabled organic supports (build plate only).
3. As can be seen from previewing slicing this, some supports intersect the second object:
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+linux-x64-GTK3

### Operating system

Arch Linux (rolling release)

### Printer model

Prusa Original i3 Mk3s

alleny87, 2023-05-09:
Same happens when adding multiple instances of a item and using auto arrange.
[1_card-2RSA-Badge-Holder-v5.zip](https://github.com/prusa3d/PrusaSlicer/files/11432934/1_card-2RSA-Badge-Holder-v5.zip)

luziato, 2024-03-30:
This still occurs in the 2.7.3 stable, but at least there is a message warning you about an error on layer 0

VorpalBlade, 2026-09-18:
I checked the attached file on 3.0.0-alpha11. The issue persists.