prusa3d/PrusaSlicer #10489: Heatset insert: Modifier for extra perimeters in one are generate nonsense perimeters in the infill.

LordSexy, 2023-05-04:
### Description of the bug

I want to make the walls of a little hole thicker for a heatset insert.
When I add a modifier (cylindrical) and change the perimeters of the modifier, I get the extra wall thickness, but also a a set of extra perimeters inside of the infill... 
[image]
See here

### Project file & How to reproduce

[Prusa bugreport.zip](https://github.com/prusa3d/PrusaSlicer/files/11395476/Prusa.bugreport.zip)
Create a 6 mm hole in some object
add a cyclindrical modifier
place it roughly in the area of your insert hole
add perimeter modifier (5-6)

Slice

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+win64-202303311519_signed

### Operating system

Windows 10

### Printer model

Anycubic mega pro. Soon Mk4

LordSexy, 2023-05-04:
Oh. Im pretty sure that I just did something wrong. Please feel free to close this asap, but please give me a hint how to manage this problem afterwards.

Cheers!

LordSexy, 2023-05-04:
Noooooooooo.  
Do you have any other suggestion on how to get 6 perimeters in exactly this position without doing something overkill?

closed 2023-05-04 by LordSexy

LordSexy, 2023-05-04:
[image]
Uuuuuugh I hate it. I set the model to specific coordinates and set the modifier to the same. Not very nice, but its working, I guess???
Anyone with a better idea is very much appreciated

foreachthing, 2023-05-04:
You can change the "Weltkoordinaten" (World Coordinates) of the modifier to "Objektkoordinaten" (Object or maybe even Part Coordinates),
Then set the modifier to X = Y = 0.