prusa3d/PrusaSlicer #10491: Need specific control over Brim material in MM prints

evilC, 2023-05-04:
**Is your feature request related to a problem? Please describe.**
When doing MM prints, there is one setting which governs which material is used for both supports and brim

Consider the following slice:
[image]
Here, the vast majority (Inc the whole of the first layer) is in the white material.
However, I need the supports in the blue material, so `Support material/raft/skirt extruder` is set to the one for the blue material.
I also want the brim to be in the white material, as it generally makes little sense to use a brim in a material other than the one it is butting up against.

**Describe the solution you'd like**
In an ideal world, brim material would be automatically assigned to the material that is being brimmed (ie brim lines would always be of the same material as the lines of the actual object that they are brimming.)  
Obviously this may not always be possible - at the very least, `brim material` should be a seperate option from `Support material/raft/skirt extruder`. Case in point - you almost always want support material to NOT stick to the object, but you surely would always want a raft TO stick to the object - so why have them both governed by one setting?  

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
Can't find any workaround that lets me do what I want

**Additional context**
None

morphias2004, 2024-04-07:
+1👍 

Having a brim forced in the support material defeats the purpose of the brim. The brim needs to print in the same material as the model.

evilC, 2024-04-07:
Nonsense. I do it all the time. 99% of my prints are Nylon + PEBA 90a, and
a brim in PEBA does not do much as the material is so soft. Even if the
bottom layer is PEBA, I still force the brim to be nylon (I sometimes have
to add an extra object to trick it into doing this) as it works so much
better

morphias2004, 2024-04-07:
> Nonsense. I do it all the time. 99% of my prints are Nylon + PEBA 90a, and a brim in PEBA does not do much as the material is so soft. Even if the bottom layer is PEBA, I still force the brim to be nylon (I sometimes have to add an extra object to trick it into doing this) as it works so much better Also, even if you were right, prusaslicer will often pick the material used for supports and use it for the brim, so in that scenario you would need to force the brim to use the material that the model itself uses
> […](#)
>

Rather than calling it nonsense, how about you supply screen captures of the settings and sliced model showing how to do it?

We can't get it to work.

evilC, 2024-04-07:
Example model:
[PrusaSlicer Issue 10491.zip](https://github.com/prusa3d/PrusaSlicer/files/14898945/PrusaSlicer.Issue.10491.zip)
Prusa XL profile (Everything default)
So "Perimeters and brim" is set to extruder 1
[image]
Result:
[image]
Now change that setting to 2:
[image]
Result: the same, it still uses extruder 1:
[image]

I even tried fiddling with this option, but it seems to do absolutely nothing:
[image]
As evidenced by the fact that if you slice, and then change it, the "Slice / Export" button still shows "Export", so as far as PS is concerned, nothing changed.

It seems that the reason is that the material selection option (this thing under the mouse cursor)
[image]
overrides the "perimiters and brim" option unless the material selection is set to "default". But in which case this pretty much renders the slicer useless unless what you are trying to do is print a single part model, but print the infill in a different material.

I think what we need is that the brim should always default to the same material as the perimiter that it touches, but we should be able to override it on a per-object basis (ie via this menu)?
[image]
That would account for a scenario such as this, where there is no one objectively correct choice, as both have equal surface area
[image]

morphias2004, 2024-04-07:
Thanks.

That's a very different scenario to having a brim and supports in the same print with the supports being done in a different material.

Try running this model and getting the brim to print in the base material whilst keeping the supports in the alternate material.

[image]
[model.zip](https://github.com/prusa3d/PrusaSlicer/files/14899109/model.zip)

morphias2004, 2024-04-07:
Yes - we need it to default to using the same materials as the extruder used for part the brim is applied to.

What we need is another option in the Multimaterial menu to allow specifically setting the extruder for the brim separately from raft, etc.

[image]

evilC, 2024-04-08:
> Thanks.
> 
> That's a very different scenario to having a brim and supports in the same print with the supports being done in a different material.
> 
> Try running this model and getting the brim to print in the base material whilst keeping the supports in the alternate material.
> 

[image]

It does this by default for me, but I can only set the "support material/raft/skirt extruder" setting to 2 if the wipe tower is off.
If I try turning the wipe tower on, I get this message
[image]
And it sets "support material/raft/skirt extruder" back to 0.

This message is absolutely infuriating - what has being soluble or not got to do with things? It really reeks of trying to overly mother the user and tell them what they do and don't want

morphias2004, 2024-08-16:
Still an issue.

sundewzer, 2024-10-23:
This is really a pain for MM prints. I have a PET part with a PLA brim. Really need an option to set the brim extruder.

tlachmann, 2024-11-11:
I`m also running into issues if I'm using soluble Material for Support Interface, becasue the printer wants to print hte rafte also in soluble Material, which doesntt make sense, even I dont want to use the Support interface on the Print plate cause doesnt support the temp of the print material bed temp.

The wired thing to me is, that it uses the Soluble Material as the first Material. 
The first Layer is printed:
1. all standard Brim with soluble Material in Extruder 5
2. then it changes to ABS Material in Extruder 3
3. Prints the Wipe Tower
4. object 1: prints support structures > perimeter > solid infill
5. same as 4. with next object

I modified the Extruders Settings and used Extruder 5 (soluble) for Material/raft intrface AND for material/raft/skirt extruder I choosed Extruder 3 (ABS)
To me that isn´t really intuitive.
[image]

[image]

evilC, 2024-11-12:
@tlachmann I have not had to do this in a while, but if memory serves, my workaround was to add a helper disc to the print and set that to the material I wanted the brim in.
For me, the material I want the brim in was always T0, dunno if that's relevant to why my trick worked

JRomer79, 2025-02-17:
I run into the Brim Extruder / Filament color issue consistently when trying to add a brim to the outside edge of Light Boxes that  I print. Due to the need to keep an inner image very sharp, I need to have the image (extruder 1) print first, and then a surrouning color (extruder 2) and then an outer rim (black/extruder 3). The primary printing is only 5 layers thick and the rim is about 1 inch. Due to how Slicer works, the brim is printed in white. As everyone in the thread knows, you cannot set the brim color explicitly, it will follow the first color printed.

Yes, I know how to clean the plate to have good adhesion. I still find I need a brim. Yes, I do make sure my filament is dry, at least 8 hours in the dryer and always stored in a sealable container with a desiccant pack. But 1 out of 2 prints will have a corner lift up. But not if I can print a brim. But I cannot have white filament printing on outside of black without creating a massive headache for cleanup.

I think several options would be usable: 
The first option, let us select the brim Extruder / Filament color.
The second option, let the brim color automatically select to the extruder/color it is printing next too. 
A third option, due to complexities of design, let us paint the brim color.

JRomer79, 2025-02-17:
> [@morphias2004](https://github.com/morphias2004), [@sundewzer](https://github.com/sundewzer)
> 
> I was able to get back the Brim in model material by changing the Gcode file, there I Changed One M104 and M109 GCode for nozzle temp to the wanted material temp and changed the T Code to the desired Tool.
> 
> After brim print, it will change the material also but to me that i can accept until the bug is fixed.

Could you share specifically how you did this? I cannot master it after a few tries.

closed 2026-09-29 by prusaslicer-bot[bot]