prusa3d/PrusaSlicer #10493: Bridge infill

jomba79, 2023-05-04:
I have a project, where the bridge infill must have properly like in the description. It works as it should in 2.4.2 version, but if you check the attached pictures, the bridge infill does not cross a few millimeters in 2.6alpha6. The object will broke at this part.
[Envelope_155x111.zip](https://github.com/prusa3d/PrusaSlicer/files/11401316/Envelope_155x111.zip)
[image]
[image]
[image]
[image]
[image]

kubispe1, 2023-05-31:
I'm sorry to announce that we have found that this issue requires more effort, so we have decided to postpone it for this release. The only possible solution currently is to manually set up the bridge angle.

jomba79, 2023-05-31:
Ok, thank youAz iPhone-omról küldve

ddacunha, 2023-09-12:
Is this related to #9999 ?

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.