prusa3d/PrusaSlicer #10494: upload to mk4 fails

greggjaskiewicz, 2023-05-04:
### Description of the bug

Trying the upload and print feature, with MK4 I just got Today. 
The resulting g-code file is about 100MB. At the very end of the upload, slicer told me that the upload failed. However the printer started printing happily. 
This is obviously not a feature that has been well tested with MK4 yet. 

### Project file & How to reproduce

download https://www.printables.com/model/463850-bulldog-low-poly-decoration
slice in 0.2 high quality layers, for mk4
Make sure your printer is configured via PrusaLink , using HTTP Digest setting
Slice
press "Send to printer"
Choose upload and print
Once the file is fully uploaded error appears - saying that there has been a failure . However, the printer will start printing happily. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+MacOS-arm64

### Operating system

macOS Ventura Version 13.3.1 (a) (Build 22E772610a)

### Printer model

mk4

greggjaskiewicz, 2023-05-05:
[image]
happens for any print.

greggjaskiewicz, 2023-05-05:
This also could just be bug in the firmware on the printer. It's running 4.6.1

kocikdav, 2023-05-09:
Hello, thank you for reporting. This bug is not caused by PrusaSlicer but by MK4 firmware. The firmware team is aware of this problem so hopefully it will be fixed in the next firmware release.

closed 2023-05-09 by kocikdav