prusa3d/PrusaSlicer #10496: Organic Supports start inside model

GeoSaffer, 2023-05-05:
### Description of the bug

Organic supports start inside model

Expected it to start outside

### Project file & How to reproduce

[Organic test.zip](https://github.com/prusa3d/PrusaSlicer/files/11406183/Organic.test.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6

### Operating system

Windows 10  (Microsoft Windows [Version 10.0.19044.2846])

### Printer model

Generic using a Ender3 mainboard

Jan-Soustruznik, 2023-05-29:
Hello, @GeoSaffer , 

In your attached 3mf file is set: "Supports: Everywhere" so the generate of supports are correct. If you want the support outside, you must use "Supports: Support on build plate only" More about the supports you can find on this link: https://help.prusa3d.com/article/support-material_1698#supports-from-build-plate-only

I close issue as solved, Thank you, have a nice day

closed 2023-05-29 by Jan-Soustruznik

GeoSaffer, 2023-07-12:
Did you even look at the issue, this relates to Organic Supports. I was under the impression that organic supports were to behave differently to normal supports.