prusa3d/PrusaSlicer #10497: Option to remove build plate from thumbnail and zoom to parts for G-Code thumbnail

TaakoMagnusen, 2023-05-05:
**Is your feature request related to a problem? Please describe.**
Right now when you enable thumbnails in the gcode it has the build plate as shown here
[image]

**Describe the solution you'd like**
I'd like to not have the build plate be considered for the thumbnail so it's just the printed parts on a transparent background. Additionally this would allow the thumbnail to be zoomed more since it doesnt need to consider the build plate for the thumbnial graphic.

As an example, this is what the preview looks like without the build plate (image taken from https://docs.fluidd.xyz/features/thumbnails), i believe this thumbnail was generated from Cura.

[image]

**Describe how it would work**
In the g-code thumbnail settings, provide a checkbox for "Exclude build plate from thumbnail". When enabled it will generate a thumbnail without showing the buildplate, and zoom the parts to fit the thumbnail.

**Describe alternatives you've considered**
I guess i could use Cura instead of PrusaSlicer 🤣 but i don't want to.

I believe there are external g-code thumbnail generators out there but then it becomes a multi-step process to generate the thumbnail externally and embed into your non-thumbnail g-code file.

Jan-Soustruznik, 2023-05-19:
SPE-1715

sabaatworld, 2023-06-10:
Any chances this could be prioritized?

ChaosBlades, 2023-07-17:
I'm trying to move from SuperSlicer and these thumbnails are so ugly lol. Not only are they hard to look at with the bed in the thumbnail. They are [resolution limited](https://github.com/prusa3d/PrusaSlicer/issues/10970).
SuperSlicer vs PrusaSlicer
[image]

[image]

Jan-Soustruznik, 2024-02-22:
FR closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8038

closed 2024-02-22 by Jan-Soustruznik