prusa3d/PrusaSlicer #10504: Loading screen stuck

powtac, 2023-05-06:
### Description of the bug

I want to start Prusa Slicer 2.5.2+win64 on Windows 11 again but it stops at the loading screen where it says "Vorbereiten der Einstellungs-Registerkarten..." (german). 

[image]

I already reinstalled it and even tested the 2.6 alpha6 (with and without importing the old config) but all show the same behaviour and stuck at the loading screen.

When I start the prusa-slicer-console.exe from the Prusa3D directory it says
[2023-05-06 18:28:18.443293] [0x0000579c] [trace]   Initializing StaticPrintConfigs
But still stops at the same loading screen with the same message. 

### Project file & How to reproduce

I can not access the error log. 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2+win64

### Operating system

Windows 11

### Printer model

Mini

powtac, 2023-05-06:
After a reboot of Windows 11 it works again! But I got this note

[image]

Jan-Soustruznik, 2023-05-29:
I think that we can close this issue as solved. 

This is known issue, for more info please visit the mentioned issues. 

> After a reboot of Windows 11 it works again! But I got this note
> 
> [image]

closed 2023-05-29 by Jan-Soustruznik