prusa3d/PrusaSlicer #10508: Overhang perimeter blobs

Kaelum, 2023-05-07:
### Description of the bug

While attempting to print the Voron Design Cube my printer would abort with the following error at layer 29 no matter what chances I made to the Print profile:
```
Move exceeds maximum extrusion (11.310mm^2 vs 0.640mm^2)
See the 'max_extrude_cross_section' config option for details
```
Since the `11.310mm^2` number is so high, the only possibility is bad slicing, so I looked at the code and saw this:
```
;TYPE:Overhang perimeter
;WIDTH:4
;HEIGHT:4
G1 F95.493
G1 X260.323 Y264.916 E8.96522
G1 E-1 F2400
G1 Z6 F9600
G1 X252.543 Y242.8
G1 Z5.8
G1 E1 F2400
```
Which is definitely the offending code.  Here is a picture of what was sliced, and it is the `blue blobs` that are the issue:

[image]

I couldn't find any setting for "Overhand perimeter", which I have never seen before, and is the problem.

### Project file & How to reproduce

Attempt to print the "Voron Design Cube v7" with 0.2mm layer heights, and 100% infill (doing this as I found another bug), and you will see these blobs.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+win64

### Operating system

Windows 10

### Printer model

Rat Rig V-core 3.1 500 & Enter 5 Plus

neophyl, 2023-05-07:
Without an attached project file anything is just a guess but you can check and turn off the thick bridges option of it’s on.

Kaelum, 2023-05-08:
@neophyl you were close, it was the `Detect bridging perimeters` that was causing the issue.  If I turn off `Thick bridges`, I can't even slice it.  I get a `Invalid spacing supplied to Flow::with_spacing()` error, and it looks like the print is hollowed out.  I'll add a project file to the original post.

closed 2026-09-29 by prusaslicer-bot[bot]