prusa3d/PrusaSlicer #10510: unicode

yahbluez, 2023-05-07:
### Description of the bug

Entering unicode in text is not able.

[image]

### Project file & How to reproduce

try to add unicode text to a model.

The expression input field did not even accept the start like "ctrl-shift-u" or the c&p of a unicode character.

PrusaSlicer
Version:   2.6.0-alpha6+linux-x64-GTK3
Build:     PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3-202303311518

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.1.25-1-MANJARO x86_64
Total RAM size [MB]: 33,339MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       Intel
Renderer:     Mesa Intel(R) UHD Graphics 630 (CFL GT2)
GLSL version: 4.60.0
Textures compression:       Enabled

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3-202303311518

### Operating system

manjaro linux 

### Printer model

any printer

Jony01, 2023-06-15:
Could you provide: 
1. font you use 
2. text you write

In my opinion, Copy and Paste (you mentioned as `c&p`) is working. Please describe it more deeply what is not working on your OS with copy and paste.

yahbluez, 2023-08-12:
Sorry for the late answer.

Start the Prusaslicer
right click <add shape TEXT>
try to enter any unicode character.

On a linux system i would type <sctrl><shift><u>  four numbers

That did not work the **input field did not accept the keyboard sequence** to start entering unicode at all.

It does not matter what code nor what font i chose. 

c&p

As described and screenshotted in the issue, if a try to enter a unicode charater by C&P the input field gets a "?" and the shown error message pops out.

Jony01, 2023-08-29:
I can't reproduce your issue.
[image]

closed 2023-08-29 by Jony01

yahbluez, 2023-08-29:
Tested now with rc1.

It did still not work.

How to reproduce:

- You need to work on a linux system, no matter what linux.
- entering unicode is done by typing "ctrl-shift-u <4digits>"
- doing that in the input field has no effect the unicode starting sequence "ctrl-shift-u" sequenze would just be ignored and you see only the 4 digits but not the corresponding character.

♥  this heart is made by typing "ctrl-shift-u 2665" 
This is the common way to type in unicode chars that do not have a corresponding key on any keyboard.
This is different from using a character like §.

When i try to C&P, for example the heart symbol above, the ctrl-v  did nothing in the embossed text field, it is just ignored.

If i try to C&P the character into the right tree it occurs but get deleted again if i hit return or take the focus away from the input field in the tree.

It is not an issue with my computer because the issue did not occur on any other program. 

If i can do more to help figuring out that bug, tell me.

Jony01, 2023-08-29:
Imgui do not support entering unicode by typing "ctrl-shift-u <4digits>". Sorry for that.
You still could copy and paste unicode symbols, couldn't you?
[image]
Consider that your selected font doesn't have to support all unicode symbols (e.g. heart).

yahbluez, 2023-08-29:
Yah, that is right, i need to select a font which has the character and than i can enter a unicode character by C&P.

[image]

So it is now broken down to the used library. Any idea how that will change in the future? We are in 2023 and unicode is around for a while. I wonder that i can use it at the command line but not in my favoured slicer. 

Thank you for the tip to check the used font to make C&P work.