prusa3d/PrusaSlicer #10512: paint on support blocker not blocking one tiny section

juztinh, 2023-05-07:
### Description of the bug

put a mando helmet on, then im blocking all interior supports by painting and it still leaves one tiny area that the program sends a branch to, ive tried 5 times now restarted etc, still same area,   cover, uncovered etc to no avail. i know that one branch is creating a lot of time thats not needed

### Project file & How to reproduce

[Mandalorian_Helmet.zip](https://github.com/prusa3d/PrusaSlicer/files/11415956/Mandalorian_Helmet.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 6

### Operating system

windows 10

### Printer model

lk5 longer

neophyl, 2023-05-07:
The stl is not much use.  You should attach your saved prusa slicer project file.  File>Save Project as.  The 3mf produced contains all the settings you are using as well as the painted on blockers etc.  It is basically a snapshot of your slice setup and should allow anyone to duplicate your issue.

juztinh, 2023-05-07:
I tried wS to big

juztinh, 2023-05-07:
I tried was to big

juztinh, 2023-05-07:
says 97mb after zipping

neophyl, 2023-05-07:
The reason is if I load the stl in and slice with organic supports and paint the top of the helmet with blocker I get this
[image]

As you can see its not sending support to the top.  This is why the project file is so important.

juztinh, 2023-05-07:
i will look here in just a min, however i did notice im slicing at a .12 layer and then  i changed it to .14 just to see the time difference and the support in the center went away.

juztinh, 2023-05-07:
do you know of any good lk5 profile that can be loaded in? i manually did this one following fb post spec sheet for longer specs.

neophyl, 2023-05-07:
Sorry I have never heard of a lk5.  Don't do facebook either.  
Easiest way to share profiles is to either export and then import a config bundle from one computer to another or just save the project file and load that.  You can then save any profiles loaded in to your system if you want to.  

Much easier than manually configuring things and less prone to mistakes.

juztinh, 2023-05-07:
[Mandalorian_Helmetissue.zip](https://github.com/prusa3d/PrusaSlicer/files/11416153/Mandalorian_Helmetissue.zip)

juztinh, 2023-05-07:
well, ok  its a trusted profile, either way some one has to enter a profile, so i did it manually, i import a config bundle but have to do it each time i open prusa, it will not save the config.  like i said im learning, Longer lk5 pro is a large fdm printer, follows closley to ender styles.  that file above is the file on my screen right now i clicked save project as.  and drug it over.  hope that helped some.

neophyl, 2023-05-07:
When you load a project or load a config its temporary.  Once loaded in (like I loaded your project) if I close the software then the profiles are gone.  However if I click the save icon next to the profile I want to save, so for example your lk5 Printer Profile it will then be saved to my system in a non temporary fashion.  I would have to do the same to save the Print and Filament profiles from your project if I wanted to save those permanently too.  

There's a different between Import Config/Import Config from Project and Import Config BUNDLE.  Importing a Config Bundle is a bit different.  A config bundle export saves all NON system profiles to a file.  When you import one of those it loads and saves permanently ALL the contained profiles.  Print, Filament and Printer.  Depending on the bundle contents that could easily be tens to hundreds of profiles.  They are a good way to back up all your custom profiles btw.

Back to the helmet though :)  I suspect theres a tiny facet that is folded over, or a zero size face on the model and as such you can't paint it.  However as its exceeding the angle its going to try and put support up to it.  
You can add a modifier of type blocker too, scale it up and place it to it cover the top of the helmet like so..
[image]

That does block the support in that area as its applied to every facet located inside the blocker mesh.
[image]

While I normally always use painting for support enforcing/blocking sometimes the old methods are handy, just for situations like these.

juztinh, 2023-05-08:
Awesome, thanks!  Yes, appreciate the info, always learning, trying new
things,  really appreciate your help. Will be testing different settings

nerdCopter, 2023-06-21:
another case of: generic-box support-blocker succeeded where paint-on did not.
[image]
[image]
[nerdCopter_S1V5_GoPro_Session.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11824399/nerdCopter_S1V5_GoPro_Session.3mf.zip) (just remove .zip naming, it's not double zipped)

closed 2026-09-29 by prusaslicer-bot[bot]