prusa3d/PrusaSlicer #10513: PrusaSlicer 2.6.0-alpha6 no bridges with "For support enforcers only and paint supports"

juckendesAuge, 2023-05-08:
### Description of the bug

PrusaSlicer 2.6.0-alpha6 no bridges with "For support enforcers only and paint supports"

Supported: Support enforcers only

I have painted on Support.

But then no bridges are used from layer height 9.20.

See pictures.

3mf file attached as zip.
[image]
[image]
[image]
[Eazy_Clamp_Spring_for_Brompton.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11416965/Eazy_Clamp_Spring_for_Brompton.3mf.zip)

### Project file & How to reproduce

Open the 3mf file and Check after slicing layerhigh 9.20

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3-202303311518.AppImage

### Operating system

Fedora release 37 (Thirty Seven)

### Printer model

Prusa MK3S+ 0.6mm Nozzle

neophyl, 2023-05-08:
Its because you have contact distance of your supports set to zero (soluble).  When configured like that then bridging is disabled.  In fact if you hover your mouse over the setting the tooltip popup actually explains that.

Hope you are using soluble supports or they are going to be pretty much impossible to remove.  Especially using tpu.

juckendesAuge, 2023-05-08:
But there is no sport there.
Besides, there are only 3 shifts. Therefore, I have to set to soluble 0. TPU can be removed very well with a side cutter that is smooth on one side.

neophyl, 2023-05-08:
Doesn't matter, once you set support z distance to zero it turns off bridge flow.  Therefore you get no bridge perimeters, no bridge cooling etc for the model.
Try setting a very small gap, something like 0.001.  The software will then use bridging but the supports will practically be at zero away, it wont make any difference for the actual print.

juckendesAuge, 2023-05-09:
Thank you very much for your help. :)

closed 2023-05-31 by Godrak

tg73, 2024-02-25:
@Godrak @neophyl There are now quite a few issues raised about this behaviour. I would argue that with the increased use of dissimilar materials for support (eg PETG with PLA) enabled by the XL, and in combination with paint-on support, users expect to get zero-gap support where enforced, and bridging where it's not enforced. The current behaviour is counter-intuitive. I was also confused. Maybe it made sense before support enforcers. Telling users to set the gap to some tiny value is not really a fix - it's an ugly counterintuitive kludge. Surely bridging should be enabled for unsupported areas when support is enabled only for enforcers, regardless of the gap? By what argument is the current behaviour correct/logical/useful?

tg73, 2024-02-25:
Also to add that this workaround appears to prevent a wipe tower from being used.