prusa3d/PrusaSlicer #10516: Crash after using automatic support painting

the-ress, 2023-05-08:
### Description of the bug

The application crashed unexpectedly with this crash dump:
[prusa-slicer.exe.24996.zip](https://github.com/prusa3d/PrusaSlicer/files/11422716/prusa-slicer.exe.24996.zip)

### Project file & How to reproduce

1. I had the following 3 objects added (some not hidden): [frenchie.zip](https://github.com/prusa3d/PrusaSlicer/files/11422728/frenchie.zip)
2. I had Prusament PETG and 0.1mm DETAIL preset selected for MK3S+ with default nozzle and with organic supports enabled
3. I selected `frenchiebody.stl`, pressed L
4. Then I pressed "Automatic painting"
5. The application immediately crashed

I was previously using automatic painting for the other objects (in the same instance) without issues. I also can't reproduce this crash.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+win64

### Operating system

Windows 11 22H2 (build 22621.1635)

### Printer model

Prusa MK3S+

Jan-Soustruznik, 2023-05-16:
Hello, @the-ress , 

thank you for your feedback, problem was caused if you try to generate the supports points at object which is non-printable.

Issue was created SPE-1707, we will try to fix in next release. 

Thank you, have a nice day

Jan-Soustruznik, 2023-05-18:
Problem solved in beta2. Issue was closed

closed 2023-05-18 by Jan-Soustruznik