prusa3d/PrusaSlicer #10519: Cooling for support interface layers

gryphonB-2, 2023-05-09:
This relates to the way support interfaces with the main part, especially for PETG. 
Because layer adhesion is the weak point of FDM printing, I like to minimise the use of the fan. This creates an unnecessary tradeoff between support adhesion and support Z & XY separation. 

This could be mitigated by enabling full fan for support interface layers and for external perimeters that are supported. In fact, enabling full fan for external perimeters generally should also increase the viable overhang angle.  

The image 'Fan Support Issue' I hope shows the issue. 
Settings:
Fan: Min 30%, Max 60%, Bridge 100% 
Contact Z distance: 0.2 mm, XY separation: 0.4 mm. 
Support Interface Layers: 3. 

In PETG the print will always sag on to the support and stick.
Clearly if fan were enabled at 100% for the support interface layers and external perimeters then sticking would be mitigated. This would allow a reduced separation and a much better part surface.  

In any case it would mitigate the continuing issue with support interface welding to the part. 

There is no way to do this in g-code, since the only placeholder with any context is 'layer_z' . Having placeholders that reference Feature Type might allow this to be done by editing g-code.  

The 3mf file is in MudGuards.zip

**Additional context**
Add any other context or paint drawing about the feature request here.
[image]
[image]
[MudGuards.zip](https://github.com/prusa3d/PrusaSlicer/files/11426260/MudGuards.zip)

rtyr, 2023-05-09:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9289 (and likely others). Your issue will be linked.

closed 2023-05-09 by rtyr

Tupson444, 2023-05-09:
There are comments in the generated GCODE file that can be used for find&replace, like:

;TYPE:Perimeter
;TYPE:External perimeter
;TYPE:Overhang perimeter
;TYPE:Support material
;TYPE:Support material interface
etc.

but I don't know if these can help because I don't have an idea how to use them for the part of the print _above_ support interface.

gryphonB-2, 2023-05-10:
Thanks Tupson444. Very helpful. This allows me to write a script inserting the desired g-code