prusa3d/PrusaSlicer #10525: Sloped Open Infill Stringing Mid-air

Platypidus, 2023-05-10:
### Description of the bug

When I slice parts with sloping open-infill, there are many unsupported straight lines going between the infill walls on layer changes. When printed, these lines are in mid-air and cause undesired stringing and blobs on my parts. The section of the part that is flat does not have the mid-air stringing. They appear to be some sort of unintentional artifact of how the infill generation works. It's hard to describe, but I attached several screenshots and a 3mf project file that should clearly show what I mean. It may be useful to know that Cura Slicer does not have this issue. 

[image]
[image]

### Project file & How to reproduce

As far as I can tell, any sloped object converted to open infill will have lines in midair on each layer. Open infill, meaning all top/bottom layers and perimeters removed. I tested with several infill types, and they all do it. I believe the slicer Ideamaker has the same problem, but Cura does not have this problem, as can be seen in one of the attached screenshots. The same problem is also present in older versions of PrusaSlicer.

[Open_Infill_Test.zip](https://github.com/prusa3d/PrusaSlicer/files/11437696/Open_Infill_Test.zip)
[image]
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6

### Operating system

Windows 10 Pro

### Printer model

Custom Marlin Printer

neophyl, 2023-05-10:
Just set the length of the infill anchor to zero.

Here is your project with the one setting changed
[image]

Platypidus, 2023-05-10:
Thanks for the quick response. I wasn't aware of a setting like this. Can the infill anchor be found in the regular settings? I will try this tonight.

neophyl, 2023-05-10:
Yes its a normal setting.  It is one of the Yellow ones so you must be in Advanced or Expert mode to see it.

[image]

Platypidus, 2023-05-11:
I'm not sure how I didn't find this setting before, but it definitely fixes the issue. There are no longer any undesirable infill lines. Thanks!

closed 2023-05-11 by Platypidus