prusa3d/PrusaSlicer #10527: Slicer making up layers

Thijxx, 2023-05-10:
### Description of the bug

When I slice my object the slicer makes up a few layers, they are not even in the 3D editor view.

### Project file & How to reproduce

[project file](https://github.com/prusa3d/PrusaSlicer/files/11440406/axle10b.zip)
[stl file](https://github.com/prusa3d/PrusaSlicer/files/11440410/axle10b.stl.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

Linux 6.1.26-1-MANJARO x86_64

### Printer model

MK3s

rtyr, 2023-05-10:
That is because your model is non-manifold. Repair the model.

Thijxx, 2023-05-10:
> That is because your model is non-manifold. Repair the model.

I think the model is fine.

rtyr, 2023-05-10:
> I think the model is fine.

It is not.

Repaired STL.
[repaired_axle10b.zip](https://github.com/prusa3d/PrusaSlicer/files/11441011/repaired_axle10b.zip)

rtyr, 2023-05-10:
[image]

closed 2023-05-10 by rtyr

Thijxx, 2023-05-10:
[image]

do we run the same version? I have the triangle but no 'open edges' in the Info pane

rtyr, 2023-05-10:
> o we run the same version?

Yes 2.5.2 (but Win64 vs Linux).