prusa3d/PrusaSlicer #10530: Bridges omit small features

spblat, 2023-05-10:
### Description of the bug

This is a new behavior that appeared between 2.6a3 and 2.6a6. It doesn't occur in 2.5.

Use case: Two layers of fine lettering on the print bed, with a solid bridging layer over the lettering. One image is from 2.6a3, where bridging covers all of the lettering. The other image is 2.6a6, where the bridge skips the smallest bits of lettering. The middle chunk in "R" doesn't bond to the printed part and is left behind on the print bed. (Never mind the backwards lettering in the example attachment.)
[image]
[image]

### Project file & How to reproduce

New Project. Add generic box. Add negative volume, the text stl. Adjust size and position of text STL so it's a couple of layers into the box in the bottom layers, only large enough for the lettering to be legible. Slice!

[missing bridge over small feature.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11445903/missing.bridge.over.small.feature.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6a5

### Operating system

macOS 12.6.5

### Printer model

Mk3S+

evilC, 2023-05-11:
I think I am seeing something similar
[PrusSlicer bridge issue.zip](https://github.com/prusa3d/PrusaSlicer/files/11452889/PrusSlicer.bridge.issue.zip)
See layer 108
PrusaSlicer 2.6.0-alpha6 (Note the gap where the mouse cursor is)
[image]

PrusaSlicer 2.5.0 (Different printer profile, so looks a bit different)
[image]

Cura
[image]

spblat, 2023-05-14:
This appears to be fixed in 2.6b1. Y'all are amazing 

[image]

Jan-Soustruznik, 2023-06-12:
Issue closed as solved.

closed 2023-06-12 by Jan-Soustruznik