prusa3d/PrusaSlicer #10533: Bridge direction ... "abnormality"

codefaux, 2023-05-11:
### Description of the bug

I'm printing a subframe unit with a dovetail-like guide slot. I've tried modifying the file a few different ways, but PrusaSlicer consistently fails to make the correct choice.

The piece in question is here;
[image]

Said guide slot is mirrored, and runs the entire length of the piece. It creates an overhang area of roughly 5.5mm x 330mm. One would expect any slicer to bridge this piece across the 5.5mm gap. I did. I gave it so much faith that I attempted printing it three times before I was actually present during the bridge.

PrusaSlicer is running a 332.5mm bridge a total of 18 or so times, across the 5.5mm gap THE WRONG WAY.

[image]

[image]

Upon closer inspection, the geometry which is encouraging this misbehavior is likely here -- pictured rotated 180 on X to expose underside view;
[image]

At those points, the solver has two options, and clearly selects incorrectly:
- "Route A":  (Pictured in screenshots)
-- Along the groove: ~330mm bridge, unsupported at both ends aka nearly universally unprintable on FDM
-- Across the holes: Properly supported with smallest gap, aka perfect

- "Route B": (Expected operation, bridge rotated 90 degrees to screenshots)
-- Along the groove: ~5.5mm bridge, supported at both ends, aka perfect
-- Across the holes: Unsupported on one side, total gap ~9.5mm, aka not ideal but likely very printable.

### Project file & How to reproduce

[Subframe.zip](https://github.com/prusa3d/PrusaSlicer/files/11455964/Subframe.zip)

Load and slice at 0.25mm

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+win64

### Operating system

Windows 10 64-bit

### Printer model

Ground-up custom

codefaux, 2023-05-11:
I was incorrect. I fixed that suspect geometry to make the correct choice more clear to the slicer, and PrusaSlicer still choses <s>violence lol</s> the wrong direction for the bridges. I don't really understand.

Layer before;
[image]

Layer of;
[image]

Layer after;
[image]

Part in question;
[image]

Transparent;
[image]

Project file;
[Subframe2.zip](https://github.com/prusa3d/PrusaSlicer/files/11456267/Subframe2.zip)

codefaux, 2023-05-11:
Ok so. I went back to the first project file since the second one didn't fix it and had oversized lockpin channels. I manually set the bridging angle so the bridge anchors actually connect to things, and _PrusaSlicer got mad at me._ It didn't get mad at itself when they _explicitly weren't_ connecting to things, as you can see by a lack of error message in the other screenshots.

[image]

Before;
[image]

Of;
[image]

After;
[image]

My workaround is to manually set bridging angle and ignore the warning, thankfully that doesn't break other bridges in the print. I've got no further testing to do, so I'll stop posting here until someone replies.

rcmaniac25, 2023-06-25:
Can confirm this issue too. The models https://www.printables.com/model/387217-3-layer-gift-envelopes require very specific bridging direction and until 2.6, PrusaSlicer picked the correct directions.

So first bridge it does correctly and overlaps the print below...
[image]

But then the 2nd bridge it is at the wrong angle...
[image]

This prevents folding the envelope is has been commented on a few times in that model's comments.

My thought is the new "Ensure vertical shell thickness" code in PrusaSlicer 2.6 is forcing the bridging at the incorrect angle in an effort to reduce lots of short back and forth... except that's exactly what is needed here.

Example 3mf: [Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/11859564/Issue.zip)

bramp, 2023-12-07:
I had a similar issue, the bridge would be built between two far edges, instead of via the two nearer edges.

<img width="799" alt="Screenshot 2023-12-06 at 5 08 44 pm" 
src="https://github.com/prusa3d/PrusaSlicer/assets/160627/aed2dab3-24c6-4142-98b5-54139b1ee0b7">

[image]

Trying

> turn off "extra perimeters on overhangs" or "detect bridging perimeters"" 

did not fix this for me, but I could mitigate it by setting the `Infill > Bridging Angle` to `45`.  I attached a very [simple STL](https://github.com/prusa3d/PrusaSlicer/files/13593412/bridging10533.stl.zip) showing this problem.

rcmaniac25, 2024-01-22:
Ran into this again, at least tangentially.

I went to print a large [calicat_3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14004821/calicat.zip) and noticed over 2h into the print that the body of the cat didn't really anchor the bridge infill.

[image]

As it was an 18h print, I didn't want to come back to find the bottom basically failed because of bad bridging. So to save restarting and the waste that comes from it, I cut the model up, added supports, and printed it on a different printer then attached the supports to the bed, and the print continued without issue.

It would be great if this issue was resolved because it's not fun having to remember the slicer can't bridge "perfectly" anymore (or, at least since 2.6) and needing to rush to resolve it or need to restart the print hours in.

Jan-Soustruznik, 2024-01-31:
Reported problem look similar of #10493

rcmaniac25, 2024-03-24:
So an earlier issue I had is now resolved with the new 2.7.3-alpha1 release

> ![image](https://private-user-images.githubusercontent.com/418272/248552550-edb8b3db-3658-4abe-b4a4-84481368ca84.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEyNjc2NzMsIm5iZiI6MTcxMTI2NzM3MywicGF0aCI6Ii80MTgyNzIvMjQ4NTUyNTUwLWVkYjhiM2RiLTM2NTgtNGFiZS1iNGE0LTg0NDgxMzY4Y2E4NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNFQwODAyNTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yOWU5N2NhODJjNWNhNDViYmNkZWIwODYwNWE3ZjQ2MjE4MzRjYzQwOGQzOTUzNDBkOWIxZmYxMzJjOGViODZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.84IwMNhjwnXhpyhQGKxgRwyqOhIO0JU7Edg-zvsSDTA)

But the more recent issue

> ![image](https://private-user-images.githubusercontent.com/418272/298463504-2d3caee7-639f-425a-b78c-c046767fccf2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEyNjc2NzMsIm5iZiI6MTcxMTI2NzM3MywicGF0aCI6Ii80MTgyNzIvMjk4NDYzNTA0LTJkM2NhZWU3LTYzOWYtNDI1YS1iNzhjLWMwNDY3NjdmY2NmMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNFQwODAyNTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NjFjOTNjZmRmYjY3NzA1NzA0ZjVmMTc4NWEwNzkyM2NiMzNkZTk5YWJmYjBkNGUxN2NhYzYxN2ViMzJlN2UzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.uXC6TwiGUowgRw1AuNHKaDDOlgibhWStJOkCJTfmlIM)

Still exists.

I'll make a new ticket around that since it may be different in the end. But glad to see this fixed in other areas...

rcmaniac25, 2026-09-13:
[The Calicat issue ](https://github.com/prusa3d/PrusaSlicer/issues/10533#issuecomment-1903263961) is still occurring under 3.0 alpha11