prusa3d/PrusaSlicer #10537: elegoo neptune 3 pro print bed size is wrong

goofyseeker311, 2023-05-12:
### Description of the bug

print area should be 225x225x280mm, no 200x200mm

### Project file & How to reproduce

225x225x228mm

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 6

### Operating system

win11

### Printer model

elegoo neptune 3 pro

rtyr, 2023-05-12:
> print area should be 225x225x280mm

It is like that in our bundled profiles. So I am not sure what do you mean.

Can we close the issue?

goofyseeker311, 2023-05-12:
sorry the loaded object was not auto-centered, so I might have confused things, it did fit as expected after "arrange", so if you feel so of course why not

closed 2023-05-12 by rtyr