prusa3d/PrusaSlicer #10541: Eject

bentdragon, 2023-05-13:
### Description of the bug

I Prusa 2.6.0, when I write the g-code to sd card no problem, but then when I eject the card in slicer. it works but then the usb reader as dead. Move it to a new usb port and try again, it works until I eject and then same problem. Reboot the computer and they all come back, until I eject again. It does the same to my internal SD card reader and the External SD card reader. If I write in slicer and use Windows file explorer to eject the sd card, no issues. 
Works every time. 
If I use Prusa Slicer 2.5 , I have no issues and can eject in Slicer.

### Project file & How to reproduce

no file

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S+

Jan-Soustruznik, 2023-05-15:
Hello, @bentdragon,

Thank you for our feedback. Could you check it with new released beta1, please? We did a modifications of eject function and it should works now. 

Link to beta1 page: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta1

Please let us know. 

Thank you, have a nice day

bentdragon, 2026-09-02:
Was a Prusa purist for years, but ….. we finally gave up trying to get the mk4’s to work and upgraded the lot with Bambu Lab X2D’s . Extremely happy.We tried for over a year to get Prusa’s to work. No support. And they were assembled units. You dragged it out till they were out of warrenty and then you tell us we are on our own.Well no longer a problem.

closed 2026-09-02 by HurkaF