prusa3d/PrusaSlicer #10542: Gcode viewer not showing fan speed correctly

Rburattini, 2023-05-13:
### Description of the bug

When you open a gcode in gcode viewer, the fan speed is not correct.

### Project file & How to reproduce

Create a gcode with variable fan speed an open in gcode viewer.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-Alpha5

### Operating system

MAC OS X Ventura 13.3.1

### Printer model

Creality Ender 3

rtyr, 2023-05-14:
Does it happen in current 2.6.0-beta1? If so, please provide some example g-code.

TravisWilder, 2023-05-30:
Still existiing in B3 
Gcode includes like this
M106 S45.9
Displayed is 100%

lukasmatena, 2023-06-13:
@Rburattini What language is your PrusaSlicer set to? If something else than English, could you please set it to English and check if the problem is still there? Thanks.

lukasmatena, 2023-06-13:
@Rburattini  Could you please give it a try in the following build? It is 2.6.0-rc1 plus [this commit](https://github.com/prusa3d/PrusaSlicer/commit/d492fc0efbff2758680f6ad310eb8dfd7c20c54c).

[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-linux-x64-GTK3-gd492fc0-202306130717.AppImage](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-linux-x64-GTK3-gd492fc0-202306130717.AppImage)
[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-MacOS-universal-gd492fc0ef-202306130729.dmg](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-MacOS-universal-gd492fc0ef-202306130729.dmg)
[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-win64-gd492fc0ef-202306130718.zip](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-win64-gd492fc0ef-202306130718.zip)

lukasmatena, 2023-06-14:
Should be fixed in [2.6.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc2). Closing.

closed 2023-06-14 by lukasmatena

TravisWilder, 2023-06-15:
@lukasmatena : I can confirm it is working for me now