prusa3d/PrusaSlicer #10543: Slicer display bug

eiffel31, 2023-05-13:
### Description of the bug

The problem is observed with multiple small objects. When viewing the generated slicing, some parts are not displayed.

The missing parts depends on the objects count, on the objects positions, on the feature display...
When there are missing parts, they are blinking sometimes when moving the view or when changing the displayed layers on the range selector on the right.

With some feature display, the objects are correct. For example, displaying width is correct while displaying the type (default feature) results in missing parts.
[image]
[image]
[image]

### Project file & How to reproduce

With one single object, I observed no problem.
With the attached project, you can test with multiple instances of the objects. The display bug changes depending on the count, the position...
[prusa_file.zip](https://github.com/prusa3d/PrusaSlicer/files/11469251/prusa_file.zip)
With a small position change, the problem may be more severe or disappear
[image]
...

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.2+linux-x64-GTK3

### Operating system

Linux Debian 11.6

### Printer model

X1

eiffel31, 2023-05-13:
Note: based on some experiments, there is no problem with 2 objects, display errors starts with 3 objects.
[image]
[image]

Jan-Soustruznik, 2023-05-15:
Hello, @eiffel31, 

 I can not reproduce your problem on Ubuntu - 20.04.6 LTS - GTK3 I tried change position of models, change the views, etc .. 
 
 Could you try with latest beta release - https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta1 
 I dont know if solve your issue, but some changes are made in the preview. 
 
 Anyway could you share the system info "_Help -> System info_"

eiffel31, 2023-05-15:
I just tried with 2.6.0 beta1.
The problem is still there: just opening the project and switching to the preview (^6). Then if I switch to other features, some have the problem, some have no problem.
I failed to switch from french to English, so I'll try to guess the English wording.
Element type: bug
Height: bug
Width: OK
Speed: OK
Fan Speed: bug
Temperature: bug
Volumetric flow: OK
Slice time lin: bug
Slice time log: bug
Tool: bug
Color: bug

I have a similar project file with 100 of these objects and the results are identical.

I noticed that some missing parts are flickering while moving the mouse. This may be a low level bug as mouse cursor has an impact, so here info relative to the chipset (extract from lspci):
00:02.0 VGA compatible controller: Intel Corporation TigerLake GT2 [Iris Xe Graphics] (rev 01)

The suspicious point is that it depends on the currently displayed feature, so looks like it is on the application side...

Tell me if you need more tests or version details.

TravisWilder, 2023-05-30:
For me in B3 the fan speed display Bug is still there

TravisWilder, 2023-05-30:
@eiffel31 : please see also #10331
the dynamic fan speed produces gcode like M106 S45.9, which already seems strange.
PS 2.6 shows all fan speeds with float as 100%

luco85, 2023-06-04:
Hi, i notice this issue with this project. fan speed in preview is 100% while has dynamic speed. In the export gcode there is multi M106 command whit relative % speed for fan so if I print it is ok. 
[PORTAPUNTALE RAINBOW.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11645507/PORTAPUNTALE.RAINBOW.3mf.zip)

TravisWilder, 2023-06-09:
> @eiffel31 : please see also #10331 the dynamic fan speed produces gcode like M106 S45.9, which already seems strange. PS 2.6 shows all fan speeds with float as 100%
> 
> Issues #10380 #10331 #10543 #10542 have all issues with displaying Fan Speed .. it would be really great to fix this in Beta 4 ... this is easy to reproduce

@lukasmatena - sorry to ping you directly but the issue is in there since some beta already and still in RC1
Easy to reproduce: fresh profile with MK4 (Install Wizard next next next)
Add two different size of objects and slice with generic PETG (has 30%-50% dynamic fan speed)
Change view to fan speed and it will show first layer with 0 and rest with 100% speed.
In gcode you will see M106 S38.25 - so it is including a float and this shows always as 100% in preview

lukasmatena, 2023-06-13:
@TravisWilder Thanks for the ping. I am not sure if your problem is related to this one. It may be, but I would rather move the discussion to #10331. I already posted something there.

@eiffel31 
> I failed to switch from french to English

This sounds interesting. Could you please switch to English and check if the problem is still there? Thanks.

bubnikv, 2023-06-13:
What is your operating system? Is it MacOS Ventura?

bubnikv, 2023-06-13:
It looks like the issue @eiffel31 suffers is unique and most likely it is due to a Linux OpenGL driver,
while the issue the others are reporting are connected to MacOS Ventura, right?

eiffel31, 2023-06-19:
I am not able to confirm it is a driver bug rather than an application bug.

I just tested with the 2.6.0-rc2 and I observe the same problem.

The only interesting observation I can make is that the problem depends on the feature displayed. For example, the display is fine when I display the width but fail when I display the temperature. Such a behavior let me think that the problem is more likely in the application rather than in the driver.

(Note: language switching fails due to locale not generated).

Please find attached my real user file with many objects (requires more computing time). It is easier to show the problem.
[bug_slicer_display.zip](https://github.com/prusa3d/PrusaSlicer/files/11787199/bug_slicer_display.zip)

closed 2026-09-29 by prusaslicer-bot[bot]