prusa3d/PrusaSlicer #10544: Gaps in External Perimeter when Speed is bigger than 25mm/s

xenon2008, 2023-05-13:
### Description of the bug

I get little Gaps in the external Perimeter when i set the Speed of the external Perimeter over 25mm/s!
For Example i print the Same file only with 30mm/s i got little Gaps, which i also could see in the Slicer Preview!

### Project file & How to reproduce

[image] => printed which 30mm/s on external Perimeter

[image]
Same File same Settings only external Perimeter printed with 25mm/s

Project File:
[External Perimeter Gaps Project.zip](https://github.com/prusa3d/PrusaSlicer/files/11470216/External.Perimeter.Gaps.Project.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+win64-202303311519_signed

### Operating system

Windows 10 x64

### Printer model

MK3S+ Clone

marek26340, 2023-05-16:
Hi. Just to clarify, those gaps also appear on the final printed part irl?
I am using the same build of PrusaSlicer and yes, I can also see those tiny gaps in my preview aswell (my external perimeter is usually set at around 50mm/s). But they are not showing up at all when printing, so this might just be a rendering issue or something.
While yes, the gaps do concern me sometimes, but they actually are not causing any issues when actually printed...

xenon2008, 2023-05-17:
Hi,
I have really little gaps on my Prints but not at this position where it shows on the preview.
Currently i am searching why i have this little gaps/dents

xenon2008, 2023-05-18:
> @xenon2008 Perhaps your filament is wet and bubbling when getting extruded? You may diagnose that by simply listening closely to the printer while printing. You'd hear an occasional popping noise - that's water converting into steam and basically bubbling away. You might be able to fix that by increasing the printing speeds a bit - on my Prusa Mini, giving it 70mm/s on perimeters yields much better results than for ex. 32mm/s.

I have already dried it, i think it is something other..

closed 2026-09-29 by prusaslicer-bot[bot]