prusa3d/PrusaSlicer #10547: 3md file from Alpha6 is broken in Beta 1

ccatlett1984, 2023-05-14:
### Description of the bug

File fails to slice, "invalid data"
Get an "Access Violation" dialog box.

### Project file & How to reproduce

[SV06 PLUS VORTEX Mount v1.0.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11470994/SV06.PLUS.VORTEX.Mount.v1.0.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta1+win64

### Operating system

Windows 11

### Printer model

Sovol SV06 Plus

Jan-Soustruznik, 2023-05-15:
Hello, @ccatlett1984 , 

thank you for your feedback. Problem is probably caused by the organic support generate. We will check and try to fix in next release. 

Issue created: SPE-1700

ccatlett1984, 2023-05-15:
Fixed file extension in title. Will try removing organic support settings and slicing the file.

ccatlett1984, 2023-05-15:
Accidentally closed

commit a35e0131 (2023-05-17): Fixed crash with Organic supports, zero top interfaces and raft enabled.
Fixes #10547 SPE-1700
closed 2023-05-17 by lukasmatena