prusa3d/PrusaSlicer #10548: Support interface layers not showing in slice preview

morphias2004, 2023-05-14:
### Description of the bug

Regardless of the support type used, interface layers on supports appears in the momentary preview that occurs before it generates the final view with the 'Feature Type' colour break down, but they disappear in the finished slice preview.

### Project file & How to reproduce

Slice and file with supports with any number of interface layers other than zero.

[image]

[image]

Exporting the GCode and viewing it with the GCode viewer does correctly display the interface layers.

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 Beta 1

### Operating system

Windows 11 x64 22H2

### Printer model

Prusa MK3S

morphias2004, 2023-05-14:
Closing and reopening the slicer resolved the issue.

xenon2008, 2023-05-14:
> Closing and reopening the slicer resolved the issue.

Not by me :(

closed 2026-09-29 by prusaslicer-bot[bot]