prusa3d/PrusaSlicer #10549: 2.6-Beta 1 crashes when using 2 Top interface layers in a certain angle of model

Guanabananas, 2023-05-14:
### Description of the bug

Software crashes when slicing this model with the slot facing up if you only have two top interface layers on the support. Works with other profiles. But it throws different errors every time it crashes.

Working:
[image]
[image]

Error:
[image]

Files: 
[crashingconfig.zip](https://github.com/prusa3d/PrusaSlicer/files/11471526/crashingconfig.zip)

### Project file & How to reproduce

1. Import config (0.28mm_slow)
2. Support material > Set "Top interface layers" to 2
3. Import STL
4. Slice
5. Crash

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta1

### Operating system

Windows 11 10.0.22621 version 22621

### Printer model

No

Jan-Soustruznik, 2023-05-16:
Hello, @Guanabananas , 
thank you for your feedback, problem can be reproduced. We will try to fix in next release. 
issue created: SPE-1708

FidelCapo, 2023-05-18:
This issue should be fixed in PrusaSlicer 2.6.0-beta2.
Closing.

closed 2023-05-18 by FidelCapo