prusa3d/PrusaSlicer #10550: connectors do not work

VitorFerreiraCH, 2023-05-14:
### Description of the bug

The negative part of the connector is not generated in the slicing of the object.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

BETA1

### Operating system

Edição	Windows 10 Pro Versão	22H2 Instalado em	‎26/‎10/‎2022 Compilação do SO	19045.2965 Experiência	Windows Feature Experience Pack 1000.19041.1000.0

### Printer model

Ender 3 V2

VitorFerreiraCH, 2023-05-15:
[Torso.zip](https://github.com/prusa3d/PrusaSlicer/files/11478793/Torso.zip)

lukasmatena, 2023-11-12:
@VitorFerreiraCH I am not able to reproduce it. However, we fixed a similar problem in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Could you please give it a try and let us know if the number of holes in the result matches your desires? Thanks.

closed 2026-09-29 by prusaslicer-bot[bot]