prusa3d/PrusaSlicer #10551: Version 2.6.0-beta1+win64 crashing when i slice with organic support and change the Branch Diameter.

Parijntje, 2023-05-14:
### Description of the bug

When i slice my stl file with organic support the program will crash while slicing. This is only happening when i change the Branche Diameter from 2 (Default) to 4. this was not happening in alpha 6 and will not happen with Grid support in Version 2.6.0-beta1+win64. The error that i get from Version 2.6.0-beta1+win64 is "access violation".

### Project file & How to reproduce

[Slotplaat v1.zip](https://github.com/prusa3d/PrusaSlicer/files/11471694/Slotplaat.v1.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta1+win64

### Operating system

Windows 11 Pro

### Printer model

Qidi X-CF Pro

d0m1n1qu3, 2023-05-14:
same here under Linux

Jan-Soustruznik, 2023-05-15:
Hello @Parijntje , problem is located on attached file? I cannot reproduce on my PC, could you check? 

Hello @d0m1n1qu3, Could you share the you project with issue, please? 

Thank you have a nice day

d0m1n1qu3, 2023-05-15:
here the current error message i got today :-)

[2023-05-15 14:30:08.594196] [0x00007f63357fe6c0] [error]   Had to calculate collision at radius 0 and layer 490, but precalculate was called. Performance may suffer!
Speicherzugriffsfehler (Speicherabzug geschrieben)

[BoneHorror_crash_beta1.zip](https://github.com/prusa3d/PrusaSlicer/files/11477729/BoneHorror_crash_beta1.zip)

d0m1n1qu3, 2023-05-15:
i think yes. i start PS out of a shell window and it is open since yesterday .. so i scroll up and found only this issus and many   GTK issus like this (100+)

`(prusa-slicer:260879): Gtk-CRITICAL **: 17:37:43.393: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:260879): Gtk-CRITICAL **: 17:37:43.394: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
`

`(prusa-slicer:275279): Gtk-CRITICAL **: 14:46:18.556: gtk_widget_set_allocation: assertion '_gtk_widget_get_visible (widget) || _gtk_widget_is_toplevel (widget)' failed

(prusa-slicer:275279): Gtk-CRITICAL **: 14:46:18.556: gtk_widget_set_allocation: assertion '_gtk_widget_get_visible (widget) || _gtk_widget_is_toplevel (widget)' failed
`

actually i can not slice with organic support at all .. :-O

Parijntje, 2023-05-15:
> Hello @Parijntje , problem is located on attached file? I cannot reproduce on my PC, could you check?
> 
> Hello @d0m1n1qu3, Could you share your project with issue, please?
> 
> Thank you have a nice day

Oddly enough, I don't have the problem anymore while yesterday I restarted the software several times and changed the settings several times.

Parijntje, 2023-05-15:
Okay, I'm having another crash. i have now tested and it appears that if i change the branch diameter and change the layer height from 0.08 to 0.05 it crashes. this is the file with the crash. (I checked)
[Slotplaat v1 (test).zip](https://github.com/prusa3d/PrusaSlicer/files/11479337/Slotplaat.v1.test.zip)

Jan-Soustruznik, 2023-05-16:
Hello, all, 
thank you for the help and quick response. We found that the problem is caused the same code like: https://github.com/prusa3d/PrusaSlicer/issues/10549 . So we will close this issue as duplicate. We will try to fix in next release. 

@d0m1n1qu3 https://github.com/prusa3d/PrusaSlicer/issues/10551#issuecomment-1548036078 - If I understood, slicer is freeze with this errors in shell, right? - It seems that is know GTK3 issue, that the slicer loose focus. Is happen to you often?

commit 01dec1bd (2023-05-17): Fixed crash in Organic supports.
Fixes #10551
closed 2023-05-16 by Jan-Soustruznik