prusa3d/PrusaSlicer #10553: Thumbnails are black in Fluidd preview

drakovych, 2023-05-14:
### Description of the bug

The thumbnails are black in Fluidd preview.
[image]

### Project file & How to reproduce

1. Export g-code to Fluidd
2.  Go to Jobs tab
3. All thumbnails are black

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-beta1+win64-202305121513

### Operating system

windows 11

### Printer model

ender 3 pro

Jan-Soustruznik, 2023-05-15:
Hello, @drakovych , 

Now in beta1 is thumbnails generated to G-code use the same colors as in the 3D view. Before we start to investigate .. Could you check if the extruder/filament color isn't set black color in your project files(3mf)?

drakovych, 2023-05-15:
Hello Jan,

Yes, the filament has black color. I have changed the color to another and it was changed in the preview.
Not sure this is a good idea but you are right about the issue. THX!

closed 2023-05-15 by drakovych