prusa3d/PrusaSlicer #10556: PrusaSlicer 2.6 Beta1 crashes when opening Configuration Wizard

Hovedskou, 2023-05-14:
### Description of the bug

When I open Configuration Wizard the PrusaSlicer crashes.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 Beta1

### Operating system

Win64

### Printer model

MK4

nifty1a, 2023-05-14:
I have the same error 
Edition	Windows 11 Home
Version	22H2
Installed on	‎13/‎10/‎2022
OS build	22621.1702
Experience	Windows Feature Experience Pack 1000.22641.1000.0

Jan-Soustruznik, 2023-05-15:
@Hovedskou , 

thank you for report. We will try to fix in next release. 

Issue created: SPE-1699

Evovita, 2023-05-16:
The crash - at least for me - occurs on line 162 (and same problem also noticed on line 170) of [/src/slic3r/GUI/ConfigWizard.cpp](https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/ConfigWizard.cpp) when the 
`\PrusaSlicer-beta\cache\EvoLabs.idx` cannot be found after I copied my custom Vendor profile files, including `EvoLabs.idx` file (that work in 2.5.2), into `C:\Users\{User Name}\AppData\Roaming\PrusaSlicer-beta\vendor` prior to running a debug session.

If I remove my custom Vendor profile files, the _Configuration Wizard_ window appears without error. **However I can't get my Vendor profile to appear in the _Configuration Wizard_ even with the below fix, so there is obviously more to the issue.**

**Problem**

`format` function failing because it's expecting 2 placeholders instead of just 1:

```
if (!boost::filesystem::exists(idx_path)) {
    BOOST_LOG_TRIVIAL(error) << format("Could not load bundle %1% due to missing index %1%.", id, idx_path.string());
    continue;
}
```
**Fix**

Include 2nd place holder in the format string, and swap the parameters around (I'm assuming 'bundle' refers to `idx_path.string()`):

`BOOST_LOG_TRIVIAL(error) << format("Could not load bundle %1% due to missing index %2%.", idx_path.string(), id);`

**OS specs**

Edition	Windows 11 Pro
Version	22H2
Installed on	‎21/‎09/‎2022
OS build	22621.1702
Experience	Windows Feature Experience Pack 1000.22641.1000.0

**IDE**

Microsoft Visual Studio Community 2022
Version 17.5.5
VisualStudio.17.Release/17.5.5+33627.172
Microsoft .NET Framework
Version 4.8.09032

kocikdav, 2023-05-17:
This crash is fixed in 2.6.0-beta2. Thank you for reporting.

closed 2023-05-17 by kocikdav