prusa3d/PrusaSlicer #10558: 3.6.0-beta1 crashes on specific project

miscoded, 2023-05-14:
### Description of the bug

I have a project with 4 parts (Voron parts -- don't ask) that crashes the app when I try to change the printer from a MK4 to a MK3S+.  It will also crash if I try to auto-arrange the parts (manual arrangement works fine). 
[Top Level 2 -MK3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11472616/Top.Level.2.-MK3.3mf.zip)

### Project file & How to reproduce

To reproduce, open the attached project and either:

- Change the Printer to 'Original Prusa i3 MK3S & MK3S+'. App crashes immediately.
- Click the Arrange button. Items will rearrange, but app will crash shortly after (during gcode generation?).

Probably important to state I have auto-gcode generation enabled.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer Version:   2.6.0-beta1+MacOS-arm64 Build:     PrusaSlicer-2.6.0-beta1+MacOS-arm64-202305121548  Operating System:    Macintosh System Architecture: 64 bit System Version:      macOS Ventura Version 13.3.1 (Build 22E261) Total RAM size [MB]: 137,439MB OpenGL installation GL version:   4.1.0 Profile:      Core Vendor:       Apple Renderer:     Apple M1 Ultra GLSL version: 4.10.0 Textures compression:       Enabled <details> <summary>Installed extensions:</summary> GL_APPLE_client_storage GL_APPLE_container_object_shareable GL_APPLE_flush_render GL_APPLE_rgb_422 GL_APPLE_row_bytes GL_APPLE_texture_range GL_ARB_ES2_compatibility GL_ARB_blend_func_extended GL_ARB_draw_buffers_blend GL_ARB_draw_indirect GL_ARB_explicit_attrib_location GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_occlusion_query2 GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_separate_shader_objects GL_ARB_shader_bit_encoding GL_ARB_shader_subroutine GL_ARB_shading_language_include GL_ARB_tessellation_shader GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_cube_map_array GL_ARB_texture_gather GL_ARB_texture_query_lod GL_ARB_texture_rgb10_a2ui GL_ARB_texture_storage GL_ARB_texture_swizzle GL_ARB_timer_query GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_ARB_vertex_attrib_64bit GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_viewport_array GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_texture_compression_s3tc GL_EXT_texture_filter_anisotropic GL_EXT_texture_sRGB_decode GL_NV_texture_barrier </details>

### Operating system

MacOS 13.3.1

### Printer model

Prusa MK3S+ and MK4 (crashes with either)

kubispe1, 2023-05-15:
Hi, we reproduced first issue on all platforms. We will try to fix it. Thanks
> Change the Printer to 'Original Prusa i3 MK3S & MK3S+'. App crashes immediately.

kubispe1, 2023-05-15:
there is simmilar issue: #10554

FidelCapo, 2023-05-18:
This issue should be fixed in PrusaSlicer 2.6.0-beta2.
Closing.

closed 2023-05-18 by FidelCapo