prusa3d/PrusaSlicer #10560: Ender 3 max bed size

raptorslegacy, 2023-05-14:
Hello,

Can somebody modify the bed on the ender 3 max profile?, it is set to 290 x 290 which is wrong it should be 300 x 300, as the bed is 300 by 310 , 310 becaus those 10 are were the clams are positioned. Thank you in advance

rtyr, 2023-05-15:
FYI @pmjdebruijn

Prusa-Support, 2023-05-17:
Here you can adjust the bed size on your printer profile. 

[image]

pmjdebruijn, 2023-05-17:
@raptorslegacy would you mind printing the following object (disable Skirt and don't change default model placement), and post a photo with the print still attached to the print bed.

[bed_alignment_288.zip](https://github.com/prusa3d/PrusaSlicer/files/11498370/bed_alignment_288.zip)

raptorslegacy, 2023-05-17:
Hello, sure as soon as i get home, i will do it asap :D, and post a picture here

raptorslegacy, 2023-05-17:
Here ya go
[image]

pmjdebruijn, 2023-05-19:
Ok that seems perfect, as I do want to keep that safety zone around the edge, so we'll keep it at 290x290 usable area.

That said, the bed model can use some adjustments...

@rtyr, for 2.6, you could add this:

[ender3max_bed.zip](https://github.com/prusa3d/PrusaSlicer/files/11516949/ender3max_bed.zip)

[printer_model:ENDER3MAX]
bed_model = ender3max_bed.stl
bed_texture = ender3max.svg

[printer_model:ENDER3MAXNEO]
bed_model = ender3max_bed.stl
bed_texture = ender3max.svg

[printer:*ENDER3MAX*]
bed_shape = 5x10,295x10,295x295,5x295

[printer:*ENDER3MAXNEO*]
bed_shape = 5x10,295x10,295x295,5x295

raptorslegacy, 2023-05-19:
Undestood, i bought the issue up frond as the CR10V2 has the bed set to 300 by 300 and the bed is the same x = 300 y = 310

pmjdebruijn, 2023-05-19:
Do you own a CR10V2 as well then?

raptorslegacy, 2023-05-19:
Yes, otherwise i would not have posted , they thing was that i wanted to print 2 things that are big and the cr10v2 was all good no issue and then when i swapped to ender 3 max , prusa said printer bed to small and i was....really :| but how it's the same bed, same clamps, was i missing something?

pmjdebruijn, 2023-05-19:
@rtyr hold on, let me do some more digging...

closed 2026-09-29 by prusaslicer-bot[bot]