prusa3d/PrusaSlicer #10566: Weird top layer build

gbzh01, 2023-05-15:
### Description of the bug

The top layers in version 2.6.0 alpha and beta are built not nicely on that specific file. I tried a lot to avoid and I could not. Please see my comparison between 2.5.2 and 2.6.0: It starts with the layer BEFORE the top layers start and doesn't get any better.
[image]
[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

I can't share the file here, unfortunately. I can do it upon request to a specific email address but not here for public.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 beta1

### Operating system

MacOS 13.3.1

### Printer model

Prusa MK3s

kubispe1, 2023-05-16:
Hi, please send me link or mail. 
petr.kubis@prusa3d.cz
Thanks

Jan-Soustruznik, 2023-06-07:
Hello, @gbzh01 .

I tried to print the model that you shared. Yes, you are right the inner top layer is generated differently in PrusaSlicer 2.5.2., this is caused by the improvements/implements of some old/new features. In PrusaSlicer 2.6.0 - print stability of top solid layers should be increased.

Anyway, I don't find any differences in final print, because this is inner top layer - not visible, the Top fill pattern - external, visible pattern isn't affected.

If you have different results, could you share, please?

Thank you, have a nice day

gbzh01, 2023-06-07:
Hi Jan

Thanks for checking on this. The issue appears then I print with translucent material which is the fact. Then I got ugly inner layers that are visible. It is OK when you improve this for major prints. It would just be fine when I could adjust it manually with modifiers but I could not. Can you?

Best regards,

Gregor Buehrer							
Director
___________________________

￼

		
Sputnique Creative GmbH
Industriestrasse 44
8152 Glattbrugg 
Switzerland

+41 76 586 04 83
www.SPTNQ.com

closed 2026-09-29 by prusaslicer-bot[bot]