prusa3d/PrusaSlicer #10570: Weird tree support generation

RM-Commando, 2023-05-15:
### Description of the bug

When generating tree supports, it starts normal but then goes in a bit which could cause instability for taller tree supports.

[image]
[image]
Tree support settings:
[image]

Zip of file:
[Overhang test.zip](https://github.com/prusa3d/PrusaSlicer/files/11480793/Overhang.test.zip)

### Project file & How to reproduce

Using tree supports (Settings in attached image), Make sure all of the bottom of the model has tree supports and the error occurs  

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-beta1+win64-202305121513

### Operating system

Windows 11

### Printer model

Prusa Mk3 (not s/s+)

Prusa-Support, 2023-05-17:
Can you share more pictures of the sliced model to see if how the supports generate? Would be good to have some images of more of the sliced modal to see what is happening above. 

James Dougherty
Prusa Research

RM-Commando, 2023-05-17:
Here:
[image]

Jan-Soustruznik, 2023-06-06:
SPE-1756

closed 2026-09-29 by prusaslicer-bot[bot]